ARM-DOE/pyart #732: SIGMET missing rays

Patri-meteocat, 2018-04-23:
When reading RAW sigmet files (pyart.io.read()), the data array in the PyART radar object does not contain the missing rays, not even in the form of masked values. 
It is still possible (although somewhat cumbersome) to decipher which are the rays missing using additional information (azimuth/elevation). However it  may be important to take this into account when the data processing relies on continuity. Is this a desired behaviour? Am I missing anything?

scollis, 2018-04-23:
Hey @Patri-meteocat ! great to hear from you. I am not too familiar with Missing rays. @jjhelmus Are you familiar with this? I had a quick look at readsigmet and could not see anything. Can you put a file some where? Are you looking for things like sweep transitions?

https://github.com/ARM-DOE/pyart/blob/master/pyart/io/sigmet.py#L37

jjhelmus, 2018-04-23:
Missing rays are removed in the `read_sigmet` function in the [code block on lines 170-178](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/sigmet.py#L170-L178).

scollis, 2018-04-24:
Thanks @jjhelmus we always appreciate your help! @Patri-meteocat It would help us if you let us know what you will be using the data for. Then we can make sure the routine is capturing the required information.

Patri-meteocat, 2018-04-25:
Hey Scott and Jonathan! I'm doing two things which are affected by the presence of missing rays. One is to apply the dual-PRF correction technique, which uses the local neighbourhood of the gates as reference for correction. If the missing rays are removed, then, for the gates adjacent to these rays, the algorithm is taking gates which are not neighbours for the reference. Moreover, the prf assigned to these rays might be wrong. The second thing I'm doing is writing the data into RAW formatted Sigmet files and this requires that the exact location of the missing rays is known. I will modify the read_sigmet function in my installation. However, is there a reason why these rays are removed that I am missing? Where should I put a sample sigmet file for you to check this?

scollis, 2018-05-22:
Hey @Patri-meteocat Sorry I dropped the ball on this.. You can use google, drop box or whatever.. or just email a file..

zssherman, 2024-05-15:
Closing for now, can reopen if need be.

closed 2024-05-15 by zssherman