ARM-DOE/pyart #737: Problem Plotting NEXRAD Velocity and Spectrum Width

ryangooch, 2018-05-14:
It appears that NEXRAD velocity and spectrum width aren't plotting with pyart.graph.RadarDisplay. I checked the example for this at [Py-ART Examples - Plot NEXRAD PPI](https://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_nexrad_multiple_moments.html) and it looks like the same problem is present there as well.

The figure produced is:

[image]

If you want, you can see my code to generate the error below.

```python
import pyart
import numpy as np
import matplotlib.pyplot as plt

DATA_FOLDER = '/path/to/data/'
RADAR_FILE = 'KJKL20180415_155233_V06'

def pyart_print_helper(radar, display, ax, param_dict):
    """
    Helper to plot radar data
    """
    display.plot(
        param_dict['moment'], 
        0,
        ax = ax,
        vmin = param_dict['vmin'], 
        vmax = param_dict['vmax'],
        title = param_dict['moment_text'],
        colorbar_label = param_dict['colorbar_label'],
        axislabels = param_dict['axislabels']
    )
    
    ax.set_aspect('equal')

    display.set_limits((-230,230), (-230,230), ax=ax)

if __name__ == '__main__':
    radar = pyart.io.read_nexrad_archive(DATA_FOLDER + RADAR_FILE)
    
    display = pyart.graph.RadarDisplay(radar)
    
    fig, axes = plt.subplots(nrows=3, ncols=2, figsize=(24,36))
    
    # Plot reflectivity
    param_dict_zh = {
        'moment':'reflectivity',
        'moment_text':'Horizontal Reflectivity',
        'vmin':-24,
        'vmax':64,
        'colorbar_label':'dBZ',
        'axislabels':('','North South distance from radar (km)')
    }

    # differential reflectivity
    param_dict_zdr = {
        'moment':'differential_reflectivity',
        'moment_text':'Differential Reflectivity',
        'vmin':-2,
        'vmax':4,
        'colorbar_label':'dB',
        'axislabels':('','')
    }
    
    # cross correlation ratio
    param_dict_rhohv = {
        'moment':'cross_correlation_ratio',
        'moment_text':'Cross Correlation Ratio',
        'vmin':0.875,
        'vmax':1,
        'colorbar_label':'unitless',
        'axislabels':('','North South distance from radar (km)')
    }
    
    # differential phase
    param_dict_phidp = {
        'moment':'differential_phase',
        'moment_text':'Differential Phase',
        'vmin':0,
        'vmax':360,
        'colorbar_label':'degrees',
        'axislabels':('','')
    }
    
    # velocity
    param_dict_vel = {
        'moment':'velocity',
        'moment_text':'Doppler Velocity',
        'vmin':-80,
        'vmax':80,
        'colorbar_label':'m/s',
        'axislabels':('East West distance from radar (km)','North South distance from radar (km)')
    }

    # spectrum width
    param_dict_sigma = {
        'moment':'spectrum_width',
        'moment_text':'Doppler Spectrum Width',
        'vmin':0,
        'vmax':10,
        'colorbar_label':'m/s',
        'axislabels':('East West distance from radar (km)','')
    }
    
    # make plots
    pyart_print_helper(radar, display, ax=axes[0,0], param_dict=param_dict_zh)
    pyart_print_helper(radar, display, ax=axes[0,1], param_dict=param_dict_zdr)
    pyart_print_helper(radar, display, ax=axes[1,0], param_dict=param_dict_rhohv)
    pyart_print_helper(radar, display, ax=axes[1,1], param_dict=param_dict_phidp)
    pyart_print_helper(radar, display, ax=axes[2,0], param_dict=param_dict_vel)
    pyart_print_helper(radar, display, ax=axes[2,1], param_dict=param_dict_sigma)
    plt.show()
```

scollis, 2018-05-14:
Go up one tilt/sweep

We many need to update our examples

Scott Collis

<https://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_nexrad_multiple_moments.html>

<https://user-images.githubusercontent.com/616258/40019360-3f422a64-577c-11e8-93cd-5af8b214f034.png>

ryangooch, 2018-05-14:
Thanks Scott.

That seems to have done the trick, though some of the other moments didn't plot at the higher sweep (Z_dr, rho_hv, phi_dp). I checked the radar object at sweep zero for velocity and there was data there. Is it an intended effect to not plot each moment at every tilt? Or maybe I'm missing something else, I don't use NEXRAD a lot.

-Ryan

scollis, 2018-05-14:
Yep.. Not every moment is saved at every sweep for level2 and it changes
from VCP to VCP

Scott Collis

scollis, 2018-05-18:
@zssherman Can you check to make sure our examples are not plotting the wrong tilts

zssherman, 2018-06-21:
@scollis This example on the pyart example page has velocity set to sweep 1. The
'def pyart_print_helper' function in the code provided has it defaulted at 0. Spectrum width isn't plotting in the pyart example find here https://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_nexrad_multiple_moments.html
I think the examples are fine.

zssherman, 2018-06-21:
@scollis Maybe in the example we can add a comment or docstring stated that some sweeps depending on VCP won't be saved, so possibly try different sweeps?

closed 2022-03-11 by zssherman