ARM-DOE/pyart #738: The netcdftime package is no longer included with netcdf4 1.4.0

vlouf, 2018-05-15:
Check this:
https://github.com/Unidata/netcdf4-python

The netcdftime package is no longer included with netcdf4 1.4.0, which cause a ModuleNotFoundError when trying to import pyart after updating netcdf4. 

```python
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/short/miniconda3/envs/radar/lib/python3.6/site-packages/pyart/__init__.py", line 59, in <module>
    from . import aux_io
  File "/short/miniconda3/envs/radar/lib/python3.6/site-packages/pyart/aux_io/__init__.py", line 45, in <module>
    from .sinarame_h5 import read_sinarame_h5
  File "/short/miniconda3/envs/radar/lib/python3.6/site-packages/pyart/aux_io/sinarame_h5.py", line 24, in <module>
    from netcdftime import utime
ModuleNotFoundError: No module named 'netcdftime'
```

Workaround are: downgrading to netcdf4 1.3.1 or changing line 24 in sinarame_h5.py to:
`from cftime import utime`

vlouf, 2018-05-15:
Errors keep piling up:
```python
/short/miniconda3/envs/radar/lib/python3.6/site-packages/pyart/retrieve/gate_id.py in fetch_radar_time_profile(sonde_dset, radar, time_key, height_key, nvars)
    118 
    119     radar_start = netCDF4.num2date(radar.time['data'][0], radar.time['units'])
--> 120     radar_day_start = netCDF4.datetime(radar_start.year, radar_start.month,
    121                                        radar_start.day)
    122     seconds_since_start_of_day = (radar_start - radar_day_start).seconds

AttributeError: module 'netCDF4' has no attribute 'datetime'
```

vlouf, 2018-05-15:
Just saw #736

zssherman, 2018-05-15:
@vlouf I was able to fix for linux and osx, but windows fails because for the cftime module, the conda-forge download for windows 32 is broken. And it seems for cftime and netcdf4, python 3.4 isn't supported anymore, at least from what I see on conda-forge. Added a if system == 3.4 to the code. Really all that needs to be fixed is the windows issue. I'm still looking on how to go about it.

scollis, 2018-05-15:
Can we add some code that does a try on the import for a temporary
workaround?

Scott Collis

<https://github.com/ARM-DOE/pyart/issues/738#issuecomment-389256307>, or mute
the thread

kmuehlbauer, 2018-05-15:
@vlouf @scollis Regarding conda-forge py3.4 is no longer supported as well as win32.

scollis, 2018-05-22:
We have fixed this and released Py-ART 1.9.1-Picasso

closed 2018-05-22 by scollis

jimgeorge, 2018-11-21:
The test that's performed (look for sys.version_info) is brittle - for example it does not give the expected behavior on python 2.7. A more robust method might be to use a try block around 'import cftime', catch any potential ImportErrors and import from netcdf4 or netcdftime instead. This works for me on the stock python 2.7 that comes with Centos 7.

scollis, 2018-11-21:
Thanks Jim! We will look into it 

-sent from a mobile device-

zssherman, 2018-11-30:
@jimgeorge Thanks for the suggestion, I'll submit a pull request shortly with the change.