ARM-DOE/pyart #740: MDV File Problem

toddstellies, 2018-05-18:
I run pyart in python on my linux virtualbox, from my windows pc. For some reason when I try to read an mdv file, it gives me an error. See the code below:
"
Python 2.7.14 |Anaconda, Inc.| (default, Dec  7 2017, 17:05:42) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import matplotlib.pyplot as plt
>>> import pyart

## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.
##
## If you use this software to prepare a publication, please cite:
##
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119

>>> filename = '/home/todd/RadarData/3hr/180000.mdv'
>>> radar = pyart.io.read_mdv(filename)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/todd/anaconda2/lib/python2.7/site-packages/pyart/io/mdv_radar.py", line 83, in read_mdv
    az_deg, range_km, el_deg = mdvfile._calc_geometry()
  File "/home/todd/anaconda2/lib/python2.7/site-packages/pyart/io/mdv_common.py", line 1106, in _calc_geometry
    raise NotImplementedError(message)
NotImplementedError: Unsupported projection type: 0, is MDV file in antenna coordinates?
>>> 
"
To take into consideration:

- I am relatively new to programming, enough to be considered a beginner
- A very detailed solution (like VERY detailed) would be appreciated.

Thank you

scollis, 2018-05-18:
First quick question: Are you sure this is radial (az,ele,range) data and not cartesian data (x,y,z)?

to check use http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.io.read_grid_mdv.html#pyart.io.read_grid_mdv

toddstellies, 2018-05-18:
Hi

Yes I am sure. Gives the same error whether I use read_mdv or read_grid_mdv

toddstellies, 2018-05-21:
Good day

As I have mentioned, I am a beginner ( I apologise for this). The file is available at https://drive.google.com/open?id=1MZhTfe0HtTUUQhg1wXg6XInMNrbH0Gk0

Thank you so much

toddstellies, 2018-05-22:
> Do you have radx installed? you can also try
> pyart.aux_io.read_radx
>
> if possible post the file some where for us to take a look at
>
>
> [http://cdn.sun.ac.za/100/ProductionFooter.jpg]<http://www.sun.ac.za/english/Pages/Water-crisis.aspx>
>
> The integrity and confidentiality of this email is governed by these terms. Disclaimer<http://www.sun.ac.za/emaildisclaimer>
> Die integriteit en vertroulikheid van hierdie e-pos word deur die volgende bepalings gereël. Vrywaringsklousule<http://www.sun.ac.za/emaildisclaimer>

Good day
As I have mentioned, I am a beginner ( I apologise for this). The file is available at https://drive.google.com/open?id=1MZhTfe0HtTUUQhg1wXg6XInMNrbH0Gk0
Thank you so much

scollis, 2018-05-22:
Hey @toddstellies Patience please.

scollis, 2018-05-22:
I have looked at the file. RadX returns:
```
WARNING - RadxConvert::_processMdvFile
  Cannot convert to RadxVol: ../../160000.mdv
Time for following error: 2018/05/22 12:54:07 UTC
ERROR - Mdv2NcfTrans::convertToRadxVol
  Path in use: ../../160000.mdv
  Incorrect projecttion, should be polar
  Found: Latitude/Longitude Grid (units in degrees)

ERROR - RadxPrint::_handleViaPath
  Printing file: ../../160000.mdv
ERROR - MdvRadxFile::readFromPath

ERROR - running RadxPrint

```

Py-ART io.read_mdv_grid returns

```
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-4-4dcf81f527ba> in <module>()
      1 radar_file = '/Users/scollis/Downloads/160000.mdv'
----> 2 radar = pyart.io.read_grid_mdv(radar_file)

~/anaconda3/envs/pyart-full/lib/python3.6/site-packages/pyart/io/mdv_grid.py in read_grid_mdv(filename, field_names, additional_metadata, file_field_names, exclude_fields, delay_field_loading, **kwargs)
    366     z = get_metadata('z')
    367     z['data'] = np.array(z_line, dtype='float64')
--> 368     z['units'] = zunits
    369 
    370     # metadata

UnboundLocalError: local variable 'zunits' referenced before assignment

```

So we have a file in grid coordinates with Z units Py-ART is not expecting...

Note io.read_mdv_grids doc:

```
Notes
-----
This function can only read cartesian MDV files with fields
compressed with gzip or zlib. For polar files see
:py:func:`pyart.io.read_mdv`

MDV files and Grid object are not fully interchangeable.  Specific
limitation include:

    * All fields must have the same shape and dimensions.
    * All fields must have the same projection.
    * Vlevels types must not vary.
    * Projection must not be PROJ_POLAR_RADAR (9) or PROJ_RHI_RADAR (13).
    * Correct unit in the Z axis are just availible for 'vlevel_type'
      equal to VERT_TYPE_Z(4), VERT_TYPE_ELEV(9), VERT_TYPE_AZ(17),
      VERT_TYPE_PRESSURE(3) and VERT_TYPE_THETA(7).
    * The behavior in cases of 2D data is unknown but most likely will not
      fail.
```

This could be a 2D field or an unknown vertical coordinate. This is not something the Argonne team will be fixing. I will leave the issue open if a member of the community wishes to address it (eg @gamaanderson  )

gamaanderson, 2018-05-22:
I have just proposed a pull request for a work around, setting zunits to 'unknown'

scollis, 2018-05-22:
Did you test with the file @toddstellies posted?

gamaanderson, 2018-05-22:
pyart is not a MDV reader, but one may get the raw MDV structure from 

    pyart.io.mdv_common.MdvFile(open(filename,'rb'))
and then try to extract the data from there. This will demand some work for one self, but it is a good place to start, if one has nothing else.

Athenaplot, 2018-08-10:
I have just used pyart.io.mdv_common.MdvFile()

`a = pyart.io.mdv_common.MdvFile('xxx.mdv')`
this can solve the problem that pyart.`io.read_grid_md` can't read the 2D data sometimes．

I can't make sure this way agree with any 2D data Mdv

scollis, 2020-01-22:
I am closing this.. if it is still an issue please re-open the issue or file a new one.

closed 2020-01-22 by scollis