ARM-DOE/pyart #747: cm_colorblind and pyart_HomeyerRainbow

rrniii, 2018-05-25:
Hey all,

Im trying to implement the color blind colormaps in some plots for a paper and I am running into a wall with the code. I have updated to the latesr code on conda and am now able to call pyart_HomeyerRainbow but it leaves my colour scale with a small bit of blue and then the rest is red. Not sure what I am doing wrong. The original plot and the new one are below along with the code I ued to make the plot.

Any help woudl be appreciated.
![unknown-3](https://user-images.githubusercontent.com/16181678/40557028-a
[image]
[image]

ccfca46-6046-11e8-9df3-c8c30ba55f5d.png)

#Figure 5. Example of observations made by the NXPol at 0.5 elevation on 
#Aug 3 2013 at ???? UTC showing: 
#a) the calibrated horizontal reflectivities, 
#b) calibrated horizontal reflectivities with a second trip filter applied, 
#c) classification of the NXPol observations (GC is Ground Clutter, RN is Rain, IN is insects and NO is Noise) and 
#d) the rainfall rates calculated from the corrected reflectivity classified as rain.

file='/Users/rrniii/Google_Drive/XbandDescriptionPaper/NXPol_Description_Paper_Data/20130803/ncas-mobile-xband-radar_davidstow-airfield_20130803-133213_SUR_QCR_v1.nc'
filename = file
radar = pyart.io.read(filename)  
display = pyart.graph.RadarMapDisplay(radar)

# Create figure instance 
fig = plt.figure(figsize=(14,12),dpi=300)

ZDRmin=-1.5
ZDRmax=3
dBZmin=-5
dBZmax=55

x1=.46
x2=.46
y1=.71
y2=.72
res='l'
sw=0

print radar.fixed_angle['data'][sw]
print radar.time['units']

### PPI a
#Add Second plot as axis 2 by adding a subplot
ax1=fig.add_subplot(221)
ax1.set_title('(a)', loc='left')
display.plot_ppi_map('dBuZ', sweep=sw, vmin=dBZmin, vmax=dBZmax,
                 resolution=res,
                 area_thresh=1,
                 projection='lcc',
                 lat_0=radar.latitude['data'][0],
                 lon_0=radar.longitude['data'][0],
                 min_lat=radar.latitude['data'][0]-x1,
                 max_lat=radar.latitude['data'][0]+x2,
                 min_lon=radar.longitude['data'][0]-y1,
                 max_lon=radar.longitude['data'][0]+y2,
                 cmap='pyart_HomeyerRainbow',
                 colorbar_flag=True,
                 colorbar_label='Reflectivity [dBZ]',
                 title_flag=False ) #lon_lines=np.arange(-7, 0, .5) lat_lines=np.arange(56, 59, .5) min_lon=-22, max_lon=-24, min_lat=13, max_lat=14,
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Radar Location As Dot
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],color='k')
# Add Range Rings; Not that this are workign on mapdiplay now
display.plot_range_ring(10., npts=360, line_style='k--')
display.plot_range_ring(20, npts=360, line_style='k-')
display.plot_range_ring(30., npts=360, line_style='k--')
display.plot_range_ring(40., npts=360, line_style='k-')
display.plot_range_ring(50, npts=360, line_style='k--')

### PPI b
ax1=fig.add_subplot(222)
ax1.set_title('(b)', loc='left')
display.plot_ppi_map('ZDR_ac', sweep=sw, vmin=ZDRmin, vmax=ZDRmax,
                     resolution=res,
                     area_thresh=1,
                     projection='lcc',
                     lat_0=radar.latitude['data'][0],
                     lon_0=radar.longitude['data'][0],
                     min_lat=radar.latitude['data'][0]-x1,
                     max_lat=radar.latitude['data'][0]+x2,
                     min_lon=radar.longitude['data'][0]-y1,
                     max_lon=radar.longitude['data'][0]+y2,
                     cmap='pyart_RefDiff',
                     colorbar_flag=True,
                     colorbar_label='Differential Reflectivity [dB]',
                     title_flag=False ) #lon_lines=np.arange(-7, 0, .5) lat_lines=np.arange(56, 59, .5) min_lon=-22, max_lon=-24, min_lat=13, max_lat=14,
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Radar Location As Dot
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],color='k')
# Add Range Rings; Not that this are workign on mapdiplay now
display.plot_range_ring(10., npts=360, line_style='k--')
display.plot_range_ring(20, npts=360, line_style='k-')
display.plot_range_ring(30., npts=360, line_style='k--')
display.plot_range_ring(40., npts=360, line_style='k-')
display.plot_range_ring(50, npts=360, line_style='k--')

### PPI c
ax1=fig.add_subplot(223)
ax1.set_title('(c)', loc='left')
display.plot_ppi_map('Specific_Attenuation_H', sweep=sw, vmin=0.,vmax=2,
                     resolution=res,
                     area_thresh=1,
                     projection='lcc',
                     lat_0=radar.latitude['data'][0],
                     lon_0=radar.longitude['data'][0],
                     min_lat=radar.latitude['data'][0]-x1,
                     max_lat=radar.latitude['data'][0]+x2,
                     min_lon=radar.longitude['data'][0]-y1,
                     max_lon=radar.longitude['data'][0]+y2,
                     cmap='pyart_SCook18',
                     colorbar_flag=True,
                     colorbar_label='Specific Attenuation [dB/km]',
                     title_flag=False ) #lon_lines=np.arange(-7, 0, .5) lat_lines=np.arange(56, 59, .5) min_lon=-22, max_lon=-24, min_lat=13, max_lat=14,
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Radar Location As Dot
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],color='k')
# Add Range Rings; Not that this are workign on mapdiplay now
display.plot_range_ring(10., npts=360, line_style='k--')
display.plot_range_ring(20, npts=360, line_style='k-')
display.plot_range_ring(30., npts=360, line_style='k--')
display.plot_range_ring(40., npts=360, line_style='k-')
display.plot_range_ring(50, npts=360, line_style='k--')

### PPI d
ax1=fig.add_subplot(224)
ax1.set_title('(d)', loc='left')
display.plot_ppi_map('Rainfall_ZC', sweep=sw, vmin=0, vmax=32,
                     resolution=res,
                     area_thresh=1,
                     projection='lcc',
                     lat_0=radar.latitude['data'][0],
                     lon_0=radar.longitude['data'][0],
                     min_lat=radar.latitude['data'][0]-x1,
                     max_lat=radar.latitude['data'][0]+x2,
                     min_lon=radar.longitude['data'][0]-y1,
                     max_lon=radar.longitude['data'][0]+y2,
                     cmap='pyart_RRate11',
                     colorbar_flag=True,
                     colorbar_label='Rain Rate [mm/hr]',
                     title_flag=False ) #lon_lines=np.arange(-7, 0, .5) lat_lines=np.arange(56, 59, .5) min_lon=-22, max_lon=-24, min_lat=13, max_lat=14,
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Grid to Plot
display.plot_grid_lines(ax=None, col='k', ls=':')
#Add Radar Location As Dot
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],color='k')
# Add Range Rings; Not that this are workign on mapdiplay now
display.plot_range_ring(10., npts=360, line_style='k--')
display.plot_range_ring(20, npts=360, line_style='k-')
display.plot_range_ring(30., npts=360, line_style='k--')
display.plot_range_ring(40., npts=360, line_style='k-')
display.plot_range_ring(50, npts=360, line_style='k--')

# Show Plot (Not strictly needed in ipython notebooks)
fig.tight_layout()

#fig.savefig('/Volumes/Google Drive/My Drive/XbandDescriptionPaper/PhotosforDescriptionPaper/Figure5_20130803-133213.png')

scollis, 2018-05-25:
Its not in a conda release yet.

You need to use Source

If you are using anaconda this environment should have you cooking:
https://github.com/EVS-ATMOS/environments/blob/master/radar/pyart_full.yml

Make sure you follow the instructions and run:
export COIN_INSTALL_DIR=/Users/USERNAME/CONDA_DIR/envs/pyart-full
On Mac..
Or replace with your home path

Scott Collis

<https://user-images.githubusercontent.com/16181678/40557044-bba793aa-6046-11e8-9ec1-1864be873e30.png>
[image: unknown-2]
<https://user-images.githubusercontent.com/16181678/40557046-bbe9db5c-6046-11e8-9789-87964c7c7eda.png>

zssherman, 2018-06-06:
Hi @rrniii , I was wondering if your allowed to share your data file. I was going to try to reproduce the issue using that data. I used your code, but on a different data file and the colormap was correct.

zssherman, 2019-07-24:
#856 fixes this issue.

zssherman, 2019-07-24:
Fixed, closing.

closed 2019-07-24 by zssherman