ARM-DOE/pyart #751: There is a bug in pyart/io/cfradial.py

ZhengYuRadar, 2018-06-06:
when I read the code of cfradial.py, I found that there is a value named 'radar_reciever_bandwidth' in pyart/io/cfradial.py line 62. This is wrong because you named it 'radar_receiver_bandwidth' in other file. It may cause other errors, I hope you can fix it soon.

zssherman, 2018-06-06:
Hi @ZhengYuRadar , Nice catch! I will put in a pull request to fix it.

ZhengYuRadar, 2018-06-06:
There is a another little bug in pyart/io/uf.py line 233. this code is """radar_beam_width_v = filemetadata('radar_beam_width_w')""", I guess it should be """radar_beam_width_v = filemetadata('radar_beam_width_v')""". Pyart is a excellent project which help me a lot in Radar data processing.

zssherman, 2018-06-06:
@ZhengYuRadar Absolutely correct, there is no radar_beam_width_w, the code is trying to pull metadata from filemetadata that doesn't exist. I will submit a commit to the pull request to address this issue to. Again, nice catch and I'm glad you like Py-ART!

scollis, 2018-06-07:
Perfect Issue @ZhengYuRadar .. This is a master class in how to file an issue..
Closing via #752

closed 2018-06-07 by scollis