ARM-DOE/pyart #753: Incorrect radius of influence for Barnes weighting

robwarrenwx, 2018-07-16:
The Barnes weighting function can be expressed as:

weight = exp( - (d^2) / (2 * sigma^2) )

where d is the distance between a grid point and an observation and sigma is the standard deviation. [Barnes (1964)](https://journals.ametsoc.org/doi/pdf/10.1175/1520-0450%281964%29003%3C0396%3AATFMDI%3E2.0.CO%3B2) expressed this slightly differently as:

weight = exp( - (d^2) / (4*k) )

Thus, k = (sigma^2) / 2. [Barnes (1964)](https://journals.ametsoc.org/doi/pdf/10.1175/1520-0450%281964%29003%3C0396%3AATFMDI%3E2.0.CO%3B2) defined a radius of influence (ROI), R, beyond which the weight is assumed to be zero, as

(R^2) / (4*k) = E

A value of E = 4 was recommended as this ensures that ~98 % of the influence of each observation is included. This gives R = 4*sqrt(k) = 2*sqrt(2)*sigma. The same recommendation was also made by [Pauley and Wu (1990)](https://journals.ametsoc.org/doi/pdf/10.1175/1520-0493%281990%29118%3C1145%3ATTDAAR%3E2.0.CO%3B2), although note that they again expressed the weight function slightly differently.

Currently, in _gate_to_grid_map.pyx the formulation of the weight is:

`weight = exp(-(dist2) / (2*roi2)) + 1e-5`

where `dist2` is the square of the distance between a grid point and a radar pixel and `roi2` is the square of the ROI. By this definition, R = sigma. In other words, **the ROI currently being used is a factor of 2*sqrt(2) too small**.

It is recommended that the function be reformulated such that sigma is defined by the user and the ROI is calculated following Barnes' (and Pauley and Wu's) recommendation.

ZhengYuRadar, 2018-07-16:
I guess it should be `weight = exp(-(dist2) / (0.25*roi2)) + 1e-5`.

scollis, 2018-07-16:
Thanks @robwarrenwx and @ZhengYuRadar ! 
I propose we put in another weighting function for a little while for testing as I am concerned some users may be compensating for us using too smaller radius of influence. 

We can call it BARNES_2 .. We will then do some diffs between the gridded fields to see the impact.

scollis, 2018-07-18:
@robwarrenwx @zssherman is working hard on modernizing our unit testing suite. I want to wait until that is done before submitting a PR to address this.

Here is what we will do: 
1) Create a new weighting function BARNES_2 which has the function as per the papers
2) raise a warning when users use BARNES so they know that the formulation is not correct and encourage them to use BARNES_2

After we do some testing we will determine if we want to move the code from BARNES_2 into BARNES.

When you are dealing with software that us used operationally by some agencies you are very careful changing the behavior of code.

zssherman, 2018-08-01:
@robwarrenwx I just wanted to make sure, if I'm understanding correctly, but instead of the weight for Barnes being:
weights = exp(-(dist2) / (2.0*roi2)) + 1e-5

Should be:
weights = exp(-(dist2) / (2.0 * np.sqrt(2.0) * roi)) + 1e-5

To follow Pauley and Wu's) recommendation.
And have sigma as a optional user defined argument?

I figured out how to implement the changes is the code, but I wanted to make sure I understood the formulation correctly.

robwarrenwx, 2018-08-06:
@zssherman I think the way it should be formulated is:

weights = exp(-(dist2) / (2*sigma2)) +1e-5

where sigma2 is the square of sigma, the standard deviation in the Gaussian function. The ROI could then be either specified by the user (but maybe not allowed to be less than sigma) or calculated following Pauley and Wu's (and Barnes') recommendation as:

roi = 2*sqrt(2)*sigma

As for sigma itself, it could again be either specified by the user, or computed as the maximum data spacing (i.e. the beam spacing, in meters, in either azimuth or elevation at the maximum range).

zssherman, 2018-08-13:
@robwarrenwx  Thank you that helps! I will continue working on the changes to the code. Think I almost have it.

gewitterblitz, 2018-09-14:
@robwarrenwx I agree with your suggestion. However, if we calculate the weights this way then we would not be following the exact lambda = 1.33*delta rule. It will be close though! 

Thanks for opening this issue!

zssherman, 2018-10-12:
@gewitterblitz, So @rcjackson and I have time next week and we will get Barnes 2 in with the suggest changes and corrections.

gewitterblitz, 2018-10-12:
If you have time, then you can also look into a simple modification of the one-pass Barnes filter as described in Koch [(1983)](https://journals.ametsoc.org/doi/pdf/10.1175/1520-0450%281983%29022%3C1487%3AAIBOMA%3E2.0.CO%3B2). The second pass filter is essentially using a convergence parameter called 'gamma'. So, instead of `w = exp(-r^2/k)` which is used in a one pass filter, we can use 

`w = exp(-r^2/gamma*k_0)`

where, `k_0  = (k_0*) * (L^2)`
`k_0` is kappa 
`k_0*` is kappa_star,
gamma lies between `0 and 1`

scollis, 2018-10-12:
Hey Milind. I think you should open a separate issue for this. This is a feature request 

-sent from a mobile device-

scollis, 2019-04-05:
Hey @robwarrenwx ! I think we have now addressed this through #806 
The weighting function is BARNES2... but when you use BARNES it will warn you that you should be using BARNES2.. we are keeping it like this for a small amount of time for backwards compatibility

closed 2019-04-05 by scollis