ARM-DOE/pyart #763: Add a method to apply a gatefilter to fields to mask the data

scollis, 2018-08-23:
At the moment the way I apply a gatefilter to mask (ie when I want to save) data is as follows:
```gf = pyart.filters.GateFilter(radar)
gf.exclude_below('reflectivity', 10)
nf = deepcopy(radar.fields['reflectivity'])
nf['data'] = np.ma.masked_where(gf.gate_excluded, nf['data'])
radar.add_field('filtered_refectivity', nf, replace_existing=True)```

It would be great to have a method that automates this...

something like:

radar.apply_filter(gatefilter, replace=False, fields=None)

If replace = False it will append 'filtered_'+field with areas where gatefilter.gate_exluded set to _FillValue
if replace = True it does it in-place
if fields = None it does it to all fields 

Inspired by Brenda Dolan

zssherman, 2019-05-07:
Self assigned, will look into this.

zssherman, 2022-11-08:
Working on this now