ARM-DOE/pyart #768: Possible bug in gridding or cfradial code.

zssherman, 2018-09-19:
After reading in a cfradial file. I tried to grid using:

```python
grids1 = pyart.map.grid_from_radars(
    (radar, ), grid_shape=(15, 251, 251),
    grid_limits=((0, 15000.0), (-50000, 50000), (-50000, 50000)),
    fields=['reflectivity'], gridding_algo="map_gates_to_grid",
    weighting_function='BARNES', min_radius=500.0)
grids2 = pyart.map.grid_from_radars((radar, ), grid_shape=(15, 251, 251),
    grid_limits=((0, 15000.0), (-50000, 50000), (-50000, 50000)),
    fields=['reflectivity'], gridding_algo="map_to_grid",
    weighting_function='BARNES', min_radius=500.0)
```
And receive an error:

```python
ValueError: setting an array element with a sequence.
```

The code fails using the map_to_grid algorithm. But the reason I mentioned cfradial code is that when I read in a raw sigmet file, the code works. But if I use a radar file that has been written using pyart.io.write_cfradial I encounter this error.

Will be looking into this more soon.

scollis, 2020-01-22:
@zssherman did you discover anything?

zssherman, 2020-01-22:
Yeah this should have been fixed with:
https://github.com/rtanamachi/pyart/pull/1

closed 2020-01-22 by zssherman