ARM-DOE/pyart #772: fixed_angle for NEXRAD with negative elevation

striges, 2018-10-02:
NEXRAD now enables low-level tilts scan below 0.5 in SAILS/MRLE/N0R,  even a negative deg at -0.2. pyart will wrap it to 359.8, which may cause issue when sorting tilts by elevations, and actually read odd.

```
// global attributes:
                :_NCProperties = "version=1|netcdflibversion=4.6.1|hdf5libversion=1.10.1" ;
                :comment = "359.7803,359.7803,0.4834,0.4834,0.8789,0.8789,1.3184,1.3184,359.7803,359.7803,1.8018,2.4170,3.1201,3.9990,5.0977,6.4160,7.9980,10.0195,11.9971,14.0186,16.6992,19.5117" ;
                :scan_id = 215 ;
                :site_name = "KMAX" ;
                :title = "" ;
                :original_container = "NEXRAD Level II" ;
                :source = "" ;
                :version = "1.3" ;
                :instrument_name = "" ;
                :references = "" ;
                :vcp_pattern = 215L ;
                :volume_start_time = "20181002173057" ;
                :Conventions = "CF/Radial instrument_parameters" ;
                :institution = "" ;
                :field_names = "cross_correlation_ratio, differential_reflectivity, reflectivity, differential_phase" ;
                :history = "" ;
data:

 fixed_angle = 359.7803 ;
```

striges, 2018-10-02:
I am working on a patch now as it is impacting operational run, I may send a pull request once I finish verification.

scollis, 2018-10-02:
Much appreciated!

Scott Collis

<https://github.com/ARM-DOE/pyart/issues/772#issuecomment-426378016>, or mute
the thread

nguy, 2019-01-28:
Anything new on this issue?

scollis, 2019-01-28:
Sorry! Life has been complex. Hey @jhemedin This is yours now! That's what you are doing to start the week First thing: Grab the KAMX file for 20181002173057 and take a look

nguy, 2019-01-28:
What, it's not like you've been to Alaska or something...oh wait!

FYI, KMUX has quite a few of these as well in case you need more test cases.

scollis, 2019-01-31:
I am encouraging @jhemedin to ask questions here. Please help him out as needed as he is new to the team

nguy, 2019-02-01:
@jhemedin if you want to send email, I'm happy to correspond that way as well and we can bring anything relevant back into this stream.

jhemedin, 2019-02-01:
@nguy I just have a question as to why the fixed angles repeat/2 of each one. I see the issue is that the negative tilts are being subtracted from 360 which is just a sorting issue with the fixed_angles. But then why does it goes "359.7803, 359.7803, 0.4834, 0.4834, 0.8789, 0.8789, 1.3184,  1.3184, 359.7803, 359.7803, 1.8018, 2.4170, 3.1201, 3.9990,5.0977, 6.4160, 7.9980, 10.0195, 11.9971, 14.0186, 16.6992, 19.5117". I'm not really familiar with SAILS/MRLE/N0R scan patterns.

striges, 2019-02-01:
CS and CD scans, one is surveillance scan, one is for doppler velocity

-Jingyin

nguy, 2019-02-01:
@jhemedin per @striges , the basis for this is to use dual PRFs to help eliminate "folding" in the observation. So each pair you note above is using a distinct PRF. These are often used in widespread or long (think squall line) events that have a higher probability of aliased returns. 

The above is known as the SZ-2 - you can find quite a bit online about this technique.

I _think_ [but have not confirmed] that the issue here may be due to how some angular/phase data are handled in PyART, to help in placing this on a 0 to 2*pi scale as circular statistics and unwrapping code more easily deal with this directly. If that helps.

striges, 2019-02-02:
Actually in msg31 the angle was encoded from 0-360, mathematically it is same, but it simply causes an issue if we sort sweeps by angle. I handled this case outside pyart so that’s the reason I opened this issue but never get a patch here.

-Jingyin

jhemedin, 2019-02-04:
@nguy Do you know of any days where KMUX has a negative tilt? Any of the dates/times I've looked at don't have one.

nguy, 2019-02-13:
Sorry, just remembered to look at this. According to [this site](https://www.roc.noaa.gov/WSR88D/SafetyandEnv/EAReports.aspx) KMAX, KMUX, KRGX, and KMSX have been approved to operate at negative tilt angles.

I can't find my KMUX notes when it happened, but the following file has an example from today if that is helpful
KMAX20190211_114803_V06

zssherman, 2019-04-25:
Fixed in #809 closing.

closed 2019-04-25 by zssherman