ARM-DOE/pyart #780: Add an `include_fields` argument to pyart.io.read

vlouf, 2018-11-05:
Currently the arguments exclude_fields is available in pyart.io.read() and pyart.io.read_grid().

The opposite would be nice when, for example, we only want to read the reflectivity in an heavy file containing lots of different fields, it would make the loading faster:

```python
radar = pyart.io.read(input_file, include_fields=['reflectivity'])  # Radar now only contains the reflectivity field
```

However, if you plan to switch to xarray as data model, then you probably won't need it.

scollis, 2018-11-05:
Feel free to ask @zssherman for help as well.

scollis, 2018-11-05:
I am happy to prioritize DIGR resources while @rcjackson is visiting @vlouf and Joshua

josephhardinee, 2018-11-05:
I thought PyART did lazy loading from Netcdf files which would mean no speed increase?

scollis, 2018-11-05:
@josephhardinee Its also useful for memory issues. If you know you are not going to use many of the vars you can exclude them from hard loading.. like C-SAPR2 :)

josephhardinee, 2018-11-05:
I mean I thought they were only loaded when  accessed (lazy load), or at least I thought I remember Jonathan implementing that a while back.

josephhardinee, 2018-11-05:
Or do you mean for binaries, load the fields but drop them before finalizing object which would hit memory initially, but then drop down in size after the load is complete.

rcjackson, 2018-11-05:
From what I read in the docs, pyart.io.read only does lazy loading if delay_field_loading is set to True, and that is not supported for every file format:

delay_field_loading : bool

True to delay loading of field data from the file until the ‘data’ key in a particular field dictionary is accessed. In this case the field attribute of the returned Radar object will contain LazyLoadDict objects not dict objects. Not all file types support this parameter.

Therefore, we would want something that is compatible for every file format.

closed 2018-11-13 by vlouf