ARM-DOE/pyart #782: more basemap issues.. might need to roll version back

scollis, 2018-11-07:
See https://github.com/matplotlib/basemap/issues/435
 In the CI I am getting:
```
../../../../miniconda3/envs/testenv/lib/python3.6/site-packages/pyart/graph/tests/test_gridmapdisplay.py:25: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../../miniconda3/envs/testenv/lib/python3.6/site-packages/pyart/graph/gridmapdisplay.py:212: in plot_grid
    latlon=True, **kwargs)
../../../../miniconda3/envs/testenv/lib/python3.6/site-packages/mpl_toolkits/basemap/__init__.py:546: in with_transform
    return plotfunc(self,x,y,data,*args,**kwargs)
../../../../miniconda3/envs/testenv/lib/python3.6/site-packages/mpl_toolkits/basemap/__init__.py:3437: in pcolormesh
    self._restore_hold(ax)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
self = <mpl_toolkits.basemap.Basemap object at 0x7f8c38e24a20>
ax = <matplotlib.axes._subplots.AxesSubplot object at 0x7f8c3a39a780>
    def _restore_hold(self, ax):
        if hasattr(ax, '_hold'):
>           ax._hold = self._tmp_hold
E           AttributeError: can't set attribute
../../../../miniconda3/envs/testenv/lib/python3.6/site-packages/mpl_toolkits/basemap/__init__.py:3238: AttributeError
__________________________ test_gridmapdisplay_fancy ___________________________
```

@zssherman can you please investigate..

zssherman, 2018-11-07:
@scollis Yeah I was looking into it, and it seems the new matplotlib is incompatible with basemap. So with the unit tests, we might have to just accept and until the new release, and the python 3.6 unit tests will work. I don't know if that's a good approach, but we can also in the environment yml file, point to an earlier matplotlib version in the mean time.

scollis, 2018-11-07:
Hang on for a while while I get an answer from matplotlib on time line

scollis, 2018-11-07:
New version being released tomorrow.. Chill

closed 2018-11-07 by scollis