ARM-DOE/pyart #785: nexrad_level2 bug when reading incomplete files

nguy, 2018-11-26:
I stumbled across this the other day. 
When accessing this file `2018/11/20/KVWX/KVWX20181120_174530_V06` pulled from the AWS NOAA bucket, I receive the following error output.

It is a partial scan that bombs out and I was able to trace this back through PyART workflow. When [getting the scan info](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/nexrad_archive.py#L113), this line fails with IndexError. I was able to trace this back to the fact that with scans=None, the nscans variable is used to generate a list list of len(nscans). However, some  `scan_msgs` (data fields) are empty lists and therefore are not iterable. I can bypass this by explicitly setting a scans keyword to [0]. Not quite sure where the best check would be to catch this.

zssherman, 2018-11-26:
Thanks @nguy for catching this! I'll try to figure out where a best check and catch can be implemented.

nguy, 2018-11-26:
No problem, let me know if you need more info or chat.

scollis, 2020-01-22:
@zssherman can you please reproduce and let us know your thoughts.

zssherman, 2020-01-22:
Yep sorry lost track of this, downloading the file now.

zssherman, 2020-01-22:
Hmm can't reproduce, read the file KVWX20181120_174530_V06 with no errors.... and digging deeper looks like this was fixed in #832 so this can be closed.

closed 2020-01-22 by zssherman