ARM-DOE/pyart #805: sudo -H pip install pyart fails on Ubuntu 16.04.5 LTS

henhen724, 2019-01-02:
When I run sudo -H pip install pyart I get the following error:
'''
Collecting pyart                                                                                                                                                                                                                       Using cached https://files.pythonhosted.org/packages/bd/88/813ba6b870578e337b33b8954a4ec2ed87900934007ec2a7d2338935b316/pyart-0.2.3.tar.gz                                                                                             Complete output from command python setup.py egg_info:                                                                                                                                                                               Traceback (most recent call last):                                                                                                                                                                                                     File "", line 1, in                                                                                                                                                                                                  File "/tmp/pip-install-0sunoelk/pyart/setup.py", line 139, in                                                                                                                                                                  kwargs['sources'] = prepare_sources(kwargs['sources'])                                                                                                                                                                             File "/tmp/pip-install-0sunoelk/pyart/setup.py", line 121, in prepare_sources                                                                                                                                                          return [to_string(source) for source in sources]                                                                                                                                                                                   File "/tmp/pip-install-0sunoelk/pyart/setup.py", line 121, in                                                                                                                                                                return [to_string(source) for source in sources]                                                                                                                                                                                   File "/tmp/pip-install-0sunoelk/pyart/setup.py", line 118, in to_string                                                                                                                                                                if isinstance(s, unicode):                                                                                                                                                                                                       NameError: name 'unicode' is not defined                                                                                                                                                                                                                                                                                                                                                                                                                                  ----------------------------------------                                                                                                                                                                                         Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-0sunoelk/pyart/
--
'''

henhen724, 2019-01-02:
that fixed the problem.
import pyart will still work in python itself though correct?

henhen724, 2019-01-02:
Thank you for your help

zssherman, 2019-01-02:
Yeah in Python you still will use `import pyart`. And no problem!

zssherman, 2019-04-25:
Added info in INSTALL.rst on pip install arm_pyart. Closing.

closed 2019-04-25 by zssherman