ARM-DOE/pyart #808: Adding radar.get_gate_longitude() instead of using attributes radar.gate_longitude

aschueth, 2019-01-25:
I am trying to map out the lat and lons for a single sweep. With radar.get_gate_x_y_z() I can specify the sweep and it will return the x,y, and z arrays for each bin in the sweep, however there is not a function like that for lat or lon. I have been trying to use the attributes, but it includes all the sweeps and is challenging to slice the data I want. Is there any way to add this?

scollis, 2019-01-31:
Yep. This is a good idea. At the moment the team is flat out so I will come back to this later

dara74k, 2019-01-31:
Hi,
Is it possible to be able to save the (lon, lat, radarvariable) as csv also?
Thanks

zssherman, 2019-05-15:
Has been addressed in #841 closing, and at @dara74k I know this is a late response, but I recommended looking into pandas. You can create and dataframe and add radar['latitude']['data'] the longitude and variable to a dataframe and then out the dataframe with dataframe.to_csv. There is more in the pandas documentation.

closed 2019-05-15 by zssherman