ARM-DOE/pyart #810: Very outdated gcc in conda-forge causing CyLP to fail building.

rcjackson, 2019-02-06:
When I try and build @jjhelmus's branch I get the following error:

 gcc -pthread -B /home/rjackson/anaconda3/envs/adi_env3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I./cylp/cpp -I./cylp/cy -I/home/rjackson/anaconda3/envs/adi_env3/include/coin -I/home/rjackson/anaconda3/envs/adi_env3/lib/python3.6/site-packages/numpy/core/include -I. -I/home/rjackson/anaconda3/envs/adi_env3/include/python3.6m -c cylp/cpp/IClpPrimalColumnPivotBase.cpp -o build/temp.linux-x86_64-3.6/cylp/cpp/IClpPrimalColumnPivotBase.o -w
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
In file included from /home/rjackson/anaconda3/envs/adi_env3/gcc/include/c++/cstdint:35:0,
                 from /home/rjackson/anaconda3/envs/adi_env3/include/coin/CoinTypes.hpp:15,
                 from /home/rjackson/anaconda3/envs/adi_env3/include/coin/CoinHelperFunctions.hpp:24,
                 from /home/rjackson/anaconda3/envs/adi_env3/include/coin/CoinIndexedVector.hpp:20,
                 from cylp/cpp/IClpPrimalColumnPivotBase.h:6,
                 from cylp/cpp/IClpPrimalColumnPivotBase.cpp:1:
/home/rjackson/anaconda3/envs/adi_env3/gcc/include/c++/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \

I tracked this down to the very, very old version of gcc that is in conda-forge (4.8.5). I got it to build with gcc 6.3.0, so I would *not* use the conda-forge gcc to try and build CyLP.

kmuehlbauer, 2019-02-06:
@rcjackson conda-forge recently upgraded to new compilers. Good chance that updating/new creatin the env might help.

rcjackson, 2019-02-08:
I did all of that and it still tries to use gcc 4.8.5. I wound up having it redirect to 6.3.0 which was installed on the system. I just figured I'd post here in case anyone runs into this issue and wants help.

zssherman, 2019-04-19:
Just as a note, CyLP to be used in python 3.7 needed to be updated with cython. Using a pip install git+https://github.com/zssherman/CyLP.git@py37 fixes the issue.

mvanlierwalq, 2019-09-10:
Trying this, getting similar failures. @zssherman does your CyLP only work for 3.7, or also 3.6? I'm also compiling/installing a newer gcc (centos default is 4.8.5) but it'll take a while to test that one out

zssherman, 2019-09-10:
@mvanlierwalq The main branch of CyLP should have the updated version, so my version isn't needed anymore. The main CyLP branch should be able to run on python 3.6, 3.7 etc. I would make sure you have the most recent cython and have the conda c compilers.

mvanlierwalq, 2019-09-10:
as noted on the other issue, after installing the condaforge c and cxx compilers, updating cython, and recompiling and reinstalling cylp from scratch, this works. Thanks for the help!

zssherman, 2019-10-01:
With the updated CyLP branch and mention of compilers in README and INSTALL files, I will close this issue for now.

closed 2019-10-01 by zssherman