ARM-DOE/pyart #813: Adding additional colorbar functionality

atmosmattbrewer, 2019-02-18:
Hello,
So I have never made edits to any package before but I was frustrated by the lack of changes to colorbar parameters that were in the original code, or at least from what I could figure out. Thus I decided to add a couple extra line to the radardisplay.py script under the plot_colorbar:

def plot_colorbar(self, **label_size =22,ticklabel_size = 22,fraction=0.046,** mappable=None, 		field=None, label=None,orient='vertical', cax=None, ax=None, fig=None,ticks=None, 			ticklabs=None):  
        Plot a colorbar.

        Parameters
       
        mappable : Image, ContourSet, etc.
            Image, ContourSet, etc to which the colorbar applied.  If None the
            last mappable object will be used.
        field : str
            Field to label colorbar with.
	**label_size : int
		Font size of the colorbar label. Default is 22
	ticklabel_size : int
		Font size for the tick label on the colorbar. Default is 22
	Fraction : int
		Fraction of original axes to use for colorbar. Default of 0.046 seems to work well from my testing**
        label : str
            Colorbar label.  None will use a default value from the last field
            plotted.
        orient : str
            Colorbar orientation, either 'vertical' [default] or 'horizontal'.
        cax : Axis
            Axis onto which the colorbar will be drawn.  None is also valid.
        ax : Axes
            Axis onto which the colorbar will be drawn. None is also valid.
        fig : Figure
            Figure to place colorbar on.  None will use the current figure.
        ticks : array
            Colorbar custom tick label locations.
        ticklabs : array
                Colorbar custom tick labels.
        
        if fig is None:
            fig = plt.gcf()
        if mappable is None:
            mappable = self.plots[-1]
        if label is None:
            if field is None:
                field = self.plot_vars[-1]
            label = self._get_colorbar_label(field)

        **cb = fig.colorbar(mappable, orientation=orient, ax=ax, cax=cax,fraction=fraction)
        cb.ax.tick_params(labelsize=ticklabel_size)**
        if ticks is not None:
            cb.set_ticks(ticks)
        if ticklabs:
            cb.set_ticklabels(ticklabs)
        **cb.set_label(label, fontsize = label_size)**

	

        self.cbs.append(cb)

I'm not sure how to contribute this to the source code or even if this is the best way to do it but it works well for what I needed. So I thought I'd share it here so maybe it could be added in the future.

scollis, 2019-02-18:
Thanks! We will take a look over this

scollis, 2020-01-22:
Anyone who wants to have a crack at this is welcome to!

HumphreysCarter, 2021-01-23:
I also need some expanded functionality of the colorbar, specifically pad and aspect, but I added all the colorbar properties for [matplotlib.pyplot.colorbar](https://matplotlib.org/3.3.3/api/_as_gen/matplotlib.pyplot.colorbar.html) from the matplotlib docs.

```
    def plot_colorbar(self, mappable=None, field=None, label=None,
                      orient='vertical', cax=None, ax=None, fig=None,
                      ticks=None, ticklabs=None, fraction=0.15, shrink=1.0,
                      aspect=20, pad=0.05, anchor=(0.0, 0.5), panchor=(1.0, 0.5)):
        """
        Plot a colorbar.

        Parameters
        ----------
        mappable : Image, ContourSet, etc.
            Image, ContourSet, etc to which the colorbar applied. If None the
            last mappable object will be used.
        field : str
            Field to label colorbar with.
        label : str
            Colorbar label. None will use a default value from the last field
            plotted.
        orient : str
            Colorbar orientation, either 'vertical' [default] or 'horizontal'.
        cax : Axis
            Axis onto which the colorbar will be drawn. None is also valid.
        ax : Axes
            Axis onto which the colorbar will be drawn. None is also valid.
        fig : Figure
            Figure to place colorbar on. None will use the current figure.
        ticks : array
            Colorbar custom tick label locations.
        ticklabs : array
            Colorbar custom tick labels.
        fraction : float
            Fraction of original axes to use for colorbar. default: 0.15
        shrink : float
            Fraction by which to multiply the size of the colorbar. default: 1.0
        aspect : float
            Ratio of long to short dimensions. default: 20
        pad : float
            Fraction of original axes between colorbar and new image axes. 0.05
            if vertical, 0.15 if horizontal
        anchor : (float, float)
            The anchor point of the colorbar axes. Defaults to (0.0, 0.5) if
            vertical; (0.5, 1.0) if horizontal.
        panchor : (float, float), or False
            The anchor point of the colorbar parent axes. If False, the parent
            axes' anchor will be unchanged. Defaults to (1.0, 0.5) if vertical;
            (0.5, 0.0) if horizontal.
        """

        # Update defaults based on orient
        if orient == 'horizontal' and pad == 0.05:
            pad = 0.15
        if orient == 'horizontal' and anchor == (0.0, 0.5):
            anchor = (0.5, 1.0)
        if orient == 'horizontal' and panchor == (1.0, 0.5):
            panchor = (0.5, 0.0)
            
        if fig is None:
            fig = plt.gcf()
        if mappable is None:
            mappable = self.plots[-1]
        if label is None:
            if field is None:
                field = self.plot_vars[-1]
            label = self._get_colorbar_label(field)

        cb = fig.colorbar(mappable, orientation=orient, ax=ax, cax=cax,
                         fraction=fraction, shrink=shrink, aspect=aspect,
                         pad=pad, anchor=anchor, panchor=panchor)

        if ticks is not None:
            cb.set_ticks(ticks)
        if ticklabs:
            cb.set_ticklabels(ticklabs)
        cb.set_label(label)

        self.cbs.append(cb)

```

zssherman, 2021-02-15:
Hi @HumphreysCarter, I think the added functionality is great! If you want to do a PR, I can review it? Cheers

HumphreysCarter, 2021-02-17:
@zssherman I'll give making a PR a shot, if I have the time over the next couple weeks. I also did a little playing around to add unit support with Pint: https://github.com/HumphreysCarter/pyart/commit/5e7ae23d9310ded92b9823249b9179e0635e933b. I couldn't see where that was an option before?

zssherman, 2021-02-17:
@HumphreysCarter Yeah no rush, I can help if need be, but from looking at the commit for the colorbar everything looks good to me. With Pint @scollis would we want temporarily add pint as an optional dependency? When we switch to xarray, i think pint is included in that dependency list, so what do you think of added pint as an option to plotting?

HumphreysCarter, 2021-03-12:
Hi @zssherman, sorry for the delay but I've submitted a pull request for the color bar additions.

zssherman, 2022-03-11:
Adding in #984 closing.

closed 2022-03-11 by zssherman