ARM-DOE/pyart #815: Plotting/displaying legacy NEXRAD level II data

kenpryor67, 2019-02-28:
Good evening-

I am attempting to plot a map of NEXRAD reflectivity data from Sterling, VA (KLWX) for September 24, 2001, using the following code:
import matplotlib.pyplot as plt
import pyart
import numpy as np

# open the file, create the displays and figure
filename = 'KLWX20010924_210141_file'
radar = pyart.io.read_nexrad_archive(filename)
display = pyart.graph.RadarDisplay(radar)
fig = plt.figure(figsize=(6, 6))

# plot super resolution reflectivity
ax = fig.add_subplot(111)
display.plot('reflectivity', 0, title='NEXRAD Reflectivity 2101 UTC',
             vmin=20, vmax=60, colorbar_label='', ax=ax)
display.plot_range_ring(radar.range['data'][-1]/1000., ax=ax)
display.set_limits(xlim=(-100, 100), ylim=(-100, 100), ax=ax)
plt.savefig("Z_2101.png",dpi=250,bbox_inches='tight')
plt.show()

max_lat = 40.5
min_lat = 37.5
min_lon = -78.5
max_lon = -75.5

lal = np.arange(min_lat, max_lat, .5)
lol = np.arange(min_lon, max_lon, .5)

display = pyart.graph.RadarMapDisplay(radar)
fig = plt.figure(figsize = [10,8])
display.plot_ppi_map('reflectivity', resolution = 'h',
                    vmin = -8, vmax = 64, mask_outside = False,
                    cmap = pyart.graph.cm.NWSRef, projection='merc',
                    min_lat = min_lat, min_lon = min_lon,
                    max_lat = max_lat, max_lon = max_lon,
                    lat_lines = lal, lon_lines = lol)

# Indicate the radar location with a point
display.plot_point(-77.299990, 39.399529, symbol='bs')
display.plot_point(-76.5280, 39.2190, symbol='bs')
plt.savefig("Ze_lwx_2101.png",dpi=250,bbox_inches='tight')

plt.show()

The basic reflectivity plot appears as expected, however, the PPI map is generated, but no data is plotted. Could this be a dataset format issue? Any advice or guidance would be greatly appreciated!

Thanks,
Ken

[image]
[image]

scollis, 2019-02-28:
What happens when you set the tilt to 0 in the map display? 

Also what happens when you do not specify any lat/lons to plot_ppi_map? Ie make the call as simple as possible? 
-sent from a mobile device-

kenpryor67, 2019-02-28:
Scott,
Thanks for the quick response!
I applied your suggestions, and used the following code to display the radar map. The data is now plotted, but no map boundaries appear. I'm now using a stereographic projection. What other projection would you recommend?

display = pyart.graph.RadarMapDisplay(radar)
fig = plt.figure(figsize = [10,8])
display.plot_ppi_map('reflectivity', sweep=0, resolution = 'h',
                    vmin = -8, vmax = 64, mask_outside = False,
                    cmap = pyart.graph.cm.NWSRef, projection='stere',
                    min_lat = None, min_lon = None,
                    max_lat = None, max_lon = None,
[image]

Thanks again for all your help. I'm looking forward to corresponding further on this issue.

kenpryor67, 2019-02-28:
Also- if I use a Lambert Conformal projection ('lcc'), a run time error appears:
RuntimeError: conic lat_1 = -lat_2

scollis, 2020-01-22:
Sorry we let this one lapse... @rcjackson is this a projection issue with Cartopy? if so I think gridlines in other projections is fixed now..

rcjackson, 2020-01-22:
I believe so. I've never had an issue when using PlateCarree projection.

mgrover1, 2022-11-07:
Closed by #1272

closed 2022-11-07 by mgrover1