ARM-DOE/pyart #816: read cinrad error

cycle13, 2019-02-28:
I have install the pyart module on my window and run the examples code well. There have some errors when I read cinrad data using 'plot_ppi_map' function. Is there anybody could help me out of these? Thank you so much.
   

...
display.plot_ppi_map

in scans [0, 2, 4, 5, 6, 7, 8, 9, 10] for moment REF.
  UserWarning)
altitude: <ndarray of type: float64 and shape: (1,)>
altitude_agl: None
antenna_transition: None
azimuth: <ndarray of type: float64 and shape: (4002,)>
elevation: <ndarray of type: float32 and shape: (4002,)>
fields:
        REF: <ndarray of type: float32 and shape: (4002, 1840)>
        VEL: <ndarray of type: float32 and shape: (4002, 1840)>
        SW: <ndarray of type: float32 and shape: (4002, 1840)>
fixed_angle: <ndarray of type: float32 and shape: (11,)>
instrument_parameters:
        unambiguous_range: <ndarray of type: float32 and shape: (4002,)>
        nyquist_velocity: <ndarray of type: float32 and shape: (4002,)>
latitude: <ndarray of type: float64 and shape: (1,)>
longitude: <ndarray of type: float64 and shape: (1,)>
nsweeps: 11
ngates: 1840
nrays: 4002
radar_calibration: None
range: <ndarray of type: float32 and shape: (1840,)>
scan_rate: None
scan_type: ppi
sweep_end_ray_index: <ndarray of type: int32 and shape: (11,)>
sweep_mode: <ndarray of type: |S20 and shape: (11,)>
sweep_number: <ndarray of type: int32 and shape: (11,)>
sweep_start_ray_index: <ndarray of type: int32 and shape: (11,)>
target_scan_rate: None
time: <ndarray of type: float64 and shape: (4002,)>
metadata:
        Conventions: CF/Radial instrument_parameters
        version: 1.3
        title: 
        institution: 
        references: 
        source: 
        history: 
        comment: 
        instrument_name: 
        original_container: CINRAD SA
Traceback (most recent call last):

  File "<ipython-input-70-09694f763b5c>", line 1, in <module>
    runfile('D:/works/pyart/pyart_tutorial/test.py', wdir='D:/works/pyart/pyart_tutorial')

  File "d:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 704, in runfile
    execfile(filename, namespace)

  File "d:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 108, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "D:/works/pyart/pyart_tutorial/test.py", line 119, in <module>
    mask_outside = False, cmap = 'pyart_NWSRef') #pyart.graph.cm.NWSRef)

  File "C:\Users\lsj\AppData\Roaming\Python\Python37\site-packages\pyart\graph\radarmapdisplay.py", line 270, in plot_ppi_map
    resolution=resolution, ax=ax, **kwargs)

  File "d:\Anaconda3\lib\site-packages\mpl_toolkits\basemap\__init__.py", line 753, in __init__
    llcrnrlon,llcrnrlat,urcrnrlon,urcrnrlat = _choosecorners(width,height,**projparams)

  File "d:\Anaconda3\lib\site-packages\mpl_toolkits\basemap\__init__.py", line 5118, in _choosecorners
    p = pyproj.Proj(kwargs)

  File "d:\Anaconda3\lib\site-packages\pyproj\__init__.py", line 362, in __new__
    return _proj.Proj.__new__(self, projstring)

  File "_proj.pyx", line 129, in _proj.Proj.__cinit__

RuntimeError: b'conic lat_1 = -lat_2'
...

scollis, 2019-04-02:
Can you make the file available?

cycle13, 2019-04-03:
still in puzzle

zssherman, 2019-04-15:
@cycle13 Can you try Cartopy instead of basemap? Not sure which pyart version you have but if it is older then the current source code (we will have a release very soon to have cartopy as default), can you try:
display = pyart.graph.RadarMapDisplayCartopy(radar)
and then:
display.plot_ppi_map
If you have the most recent code, avoid RadarMapDisplayBasemap for now.

cycle13, 2019-04-21:
Ok, let me try.
Thank you so much for your kindness help.

zssherman, 2020-03-26:
@cycle13 I'm going to close this, let me know if you still run into issues.

closed 2020-03-26 by zssherman