ARM-DOE/pyart #819: Possible bug plotting CSAPR-2 HSRHIs

swnesbitt, 2019-04-02:
In plotting CSAPR-2 HSRHIs from CACTI, I am having issues with plotting RHIs at 90° azimuth with `display.plot_rhi()`.  This behavior is repeatable amongst many files, but only at this azimuth angle.  Seems like a geometry issue.  
Example is here:
[image]
Interestingly, `radar.get_gate_x_y_z(sweep=3)` seems to extract ok values:
[image]
I'll keep digging...

scollis, 2019-04-02:
Thanks Steve. Well look into it 

-sent from a mobile device-

swnesbitt, 2019-04-02:
Yeah, the calculation for the x-axis value in the function `plot_rhi()` has a bug:

[image]

scollis, 2019-04-02:
What's strange is that it is working for some elevations past 90 degrees where sign(y) = -1.0

Can you plot the elevations for that azimuth?

swnesbitt, 2019-04-02:
Ask and you shall receive
[image]

scollis, 2019-04-02:
grabbing data

scollis, 2019-04-02:
ok @swnesbitt I have the dodge to end all dodges why we patch this:
```
radar_file = '/data/CACTI/csapr2/corcsapr2cfrhsrhiM1.a1.20181126.120715.nc'  
radar = pyart.io.read(radar_file)
radar.azimuth['data'] += .01
my_d = pyart.graph.RadarDisplay(radar)
my_d.plot_rhi('reflectivity', 3)
```

scollis, 2019-04-02:
@zssherman please add a line that, when abs(radar.fixed_angle['data'][sweep] == 90) (or y.all == 0) use sign(x)

scollis, 2019-04-02:
Thank you @swnesbitt This was a nice find. Old code is, old code..

scollis, 2019-04-03:
Closing via #820 
@swnesbitt if you install from source this should fix things

closed 2019-04-03 by scollis