ARM-DOE/pyart #823: ENH: Switch from disutils to setuptools

zssherman, 2019-04-08:
We planned on switching from disutils to setuptools for some time, but this is just to keep track and reiterate some of the possible benefits:

Can have it compile c and not have to include pyx files as mentioned in #759
Can have install_requires set, which will probably fix the current issue with appveyor.
From what I'm seeing, we can remove the setup.py files in the sub directories.

zssherman, 2022-03-14:
This would also solve issues with requirement file installs. Will be trying to tackle this soon.

mgrover1, 2022-03-16:
FYI, looking at the [Cython docs](https://cython.readthedocs.io/en/latest/src/quickstart/build.html), here is the recommended snippet to use in the `setup.py` file using setuptools, where we insert a list of all `.pyx` files.

```python
from setuptools import setup
from Cython.Build import cythonize

setup(
    name='Hello world app',
    ext_modules=cythonize("hello.pyx"),
    zip_safe=False,
)
```

closed 2022-04-06 by mgrover1