ARM-DOE/pyart #825: Add PyART to Python 3 Statement or otherwise clarify support timeline

CAM-Gerlach, 2019-04-15:
Three of the project's four core dependencies, Numpy, SciPy, and Matplotlib, have all [committed ](https://python3statement.org/)to drop Python 2 support completely on or before January 1, 2020, and Unidata, the parent org of the the fourth (NetCDF4) has [stated](https://www.unidata.ucar.edu/blogs/developer/entry/timeline-for-dropping-support-for) the same. Furthermore, Pytest, the project's testing framework, has also made the same commitment.

Therefore, as an important package used in the atmospheric science community, I suggest [adding](https://github.com/python3statement/python3statement.github.io) the project to the [Python3Statement](https://python3statement.org/), like most of the PyData stack and [MetPy](https://github.com/python3statement/python3statement.github.io/pull/123) have done, to make it easy and clear for users to find in one central place and help accelerate Python 3 adoption across our atmospheric science community. Alternatively, if PyART has plans to break with its dependencies and support Python 2 beyond 2020, I suggest clarifying them somewhere so users are appropriately informed.

Let me know if you'd like any help or have any questions. Thanks!

zssherman, 2019-04-15:
Hi @CAM-Gerlach , we do plan on removing python 2 support soon. Definitely before 2020. I will talk to @scollis about signing the statement, I don't see it being in issue in anyway, but dependency wise, basemap really is the only package we are removing which support its geared towards cartopy anyways. We will probably being adding xarray as a dependency in the near feature as well.

scollis, 2019-04-16:
Excellent point. We have been a bit swamped lately just bug squashing.. We will do so

CAM-Gerlach, 2019-05-07:
@scollis @zssherman I opened a PR, python3statement/python3statement.github.io#214 which will add this to the list if you approve it, so you don't need to do any more work to add it. Let us know over there, thanks!

zssherman, 2019-10-01:
The statement was signed few months ago, but leaving this open until Python 2.7 is removed from conda-forge.

zssherman, 2020-01-30:
Python 2 is no longer supported, closing.

closed 2020-01-30 by zssherman