ARM-DOE/pyart #829: Airborne Plotting Issue

dstex, 2019-04-19:
There seems to be something off with how airborne (specifically NOAA TDR) data are plotted. I saw @nguy opened an issue a few years back which may be related ( #564 ), though too few details were provided to know for sure.

In the Solo display, I see the precipitation echo extends all the way out to 45 km on the right, with surface returns similarly terminating at 45 km on the left. Further, the Solo examine widget indicates that the range associated with those gates correlates with what the display is showing. 
[image]

When considering the same sweep as plotted with pyart, these features fall short of the 45 km mark (the last range ring), and not symmetrically so (the left side extends to ~42.5 km, and the right side only to ~42 km).
[image]

Has anyone else seen this kind of behavior before? I assume it may have something to do with the antenna to cartesian transform, though if so, it's not clear to me what's wrong.

nguy, 2019-05-03:
@dstex It's been a while since I worked with this data. 
Could you post a snippet of the code calls you are making?
And if possible share this file with me to test? You've used this before, correct? Do you know if this is a new behavior or you've seen it previously?

dstex, 2019-07-30:
@nguy @scollis  Of note, I've gone back and checked some of my previous plots created 1+ years ago with a different TDR dataset, and this issue was indeed present then as well (and I apparently missed it).

dstex, 2019-08-01:
@nguy @scollis I have figured out what the source of the discrepancy is between what Py-ART plots and what is displayed in soloii. Py-ART is plotting the sweep in true Cartesian coordinates while soloii plots the location of each gate based solely on antenna coordinates. For example, a gate at 30 km range from the radar at 270° azimuth is plotted at 28.225 km as the `antenna_to_cartesian` function is adjusting based on the ±20° "elevation" angle of the TDR, which soloii apparently does not do.

I can adjust how this is handled as needed in my local branch, but I wonder if the ability to plot in a partially untransformed coordinate space would be a desirable option for others?

scollis, 2019-08-01:
You are a great judge of this. Is it used in the community? If yes a PR adding the option would be welcomed 

-sent from a mobile device-

dstex, 2019-08-01:
I think that most consumers of TDR/ELDORA data are likely quite used to the soloii/3 displays, and would expect (as I did) that Py-ART plots behave in the same way. I'll see if I can come up with a simple way to adjust this in Py-ART and add an argument to `radardisplay_airborne.plot` to enable it.

nguy, 2019-08-01:
Just trying to understand here, are they essentially plotting along two different axes then? Solo along the field defined by the "elevation" angle and PyART along aircraft longitudinal axis?

dstex, 2019-08-01:
Yes, plotting along different axes (I think). Essentially, solo plots along an axis perpendicular to the flight track. PyART instead accounts for the fore/aft 20° tilt of the antenna and maps the gates from that plane onto the plane perpendicular to the flight track. This results in an apparent compression of the data towards the aircraft such that gates are not plotted at the locations given by their range from the radar. Does that clear it up at all?

nguy, 2019-08-01:
This is really interesting. So just wondering in the back of my head why solo does not take into account the fore/aft tilt in it's projection calculations? Is this on purpose or an oversight?
I'd be interested to hear @mmbell thoughts on this.

swnesbitt, 2019-08-01:
Solo just plots the data with slant range, not ground distance from radar. I would say that this was the standard way to do it for a long time.

STEVE NESBITT
Professor

Department of Atmospheric Sciences
University of Illinois at Urbana-Champaign

3060 Natural History Building | M/C 104
1301 W Green St
Urbana, IL 61801

217.244.3740 | fax: 217.244.1752 | snesbitt@illinois.edu
www.atmos.illinois.edu<http://www.atmos.illinois.edu> | snesbitt.web.illinois.edu

Under the Illinois Freedom of Information Act any written communication to or from university employees regarding university business is a public record and may be subject to public disclosure.

nguy, 2019-08-01:
Welp, that would explain the differences then. So this would also affect the appearance of any ground based tilt then too if compared @swnesbitt ?

So yeah @dstex maybe look to see if you can modify the keywords. It may take a modification of the projection routine as well.

dstex, 2019-08-22:
@nguy I have a set of modifications that seem to be working, so I'll see about submitting a pull request in the near future when I have the time. I'm not positive if the appearance of ground based projections is also affected, though I expect they probably would be...

closed 2019-12-18 by dstex