ARM-DOE/pyart #839: Netcdf update breaking cfradial module.

zssherman, 2019-05-06:
Netcdf4 recently did an update and the unit tests are now failing due to a shape mismatch. If users run into this errors, downgrade nctcdf to version 1.5.0 for now. Error occurs with netCDF4 version 1.5.1.1, looking into a solution now.

```
io/grid_io.py:304: in write_grid
    _create_ncvar(grid.point_x, dset, 'point_x', ('z', 'x', 'y'))
io/cfradial.py:799: in _create_ncvar
    ncvar[:] = data[:]
netCDF4/_netCDF4.pyx:4796: in netCDF4._netCDF4.Variable.__setitem__
    ???
../../../.conda/envs/pyart_env/lib/python3.7/site-packages/netCDF4/utils.py:365: in _StartCountStride
    datashape = broadcasted_shape(shape, datashape)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

shp1 = (2, 320, 400), shp2 = (2, 400, 320)

    def broadcasted_shape(shp1, shp2):
        # determine shape of array of shp1 and shp2 broadcast against one another.
        x = np.array([1])
        # trick to define array with certain shape that doesn't allocate all the
        # memory.
        a = as_strided(x, shape=shp1, strides=[0] * len(shp1))
        b = as_strided(x, shape=shp2, strides=[0] * len(shp2))
>       return np.broadcast(a, b).shape
E       ValueError: shape mismatch: objects cannot be broadcast to a single shape
```

scollis, 2019-05-06:
Massive win for the CI here.. This is why we test.

zssherman, 2019-05-06:
Might be actually a bug in the grid io code:
if write_point_x_y_z:
304         _create_ncvar(grid.point_x, dset, 'point_x', ('z', 'x', 'y'))
305         _create_ncvar(grid.point_y, dset, 'point_y', ('z', 'x', 'y'))
306         _create_ncvar(grid.point_z, dset, 'point_z', ('z', 'x', 'y'))

I believe x and y should be flipped.

zssherman, 2019-05-06:
Also the second error looks like a bug in netCDF4 and will be fixed in [#925 netCDF4](https://github.com/Unidata/netcdf4-python/pull/925) looks like they will have the release sometime today.

zssherman, 2019-05-07:
Fixed with #840 closing.

closed 2019-05-07 by zssherman