ARM-DOE/pyart #850: Writing specific fields

bdolan44, 2019-06-12:
I think it would be very useful to be able to specify the fields to write out. Something along the lines of 

pyart.io.write(radar,fields = ['reflectivity','differential_reflectivity','velocity']")

scollis, 2019-06-12:
Great idea! Assigning @zssherman

zssherman, 2019-06-12:
Sounds good to me, I don't think it will be difficult to implement at all. I will take a look.

swnesbitt, 2019-06-12:
Chiming in here – would it be more useful to write a function that selected fields from a radar object.  This could make it easier to also do memory management, etc.

This could follow the xarray syntax, for example

radar1.sel(field=[‘reflectivity’,’velocity’])
and maybe in the future
radar1.sel(azimuth=slice([90,150]))
radar1.sel(sweep=slice(1,3))

This could easily be put into a one-liner
pyart.io.write(radar1.sel(fields = ['reflectivity','differential_reflectivity','velocity']))
or eventually (to follow pandas/xarray syntax)
radar1.sel(field=[‘reflectivity’,‘velocity’]).to_cfradial(‘file.nc’)

Just my two pennies,
-Steve

zssherman, 2019-06-12:
@swnesbitt I do like that idea, and now with xarray being used for the grids, its already a dependency. I'll see what I can code up.

bdolan44, 2019-06-13:
Yes, 1-up on @swnesbitt's comment. That would be fantastic.

zssherman, 2022-11-08:
Completed in #1319 , closing.

closed 2022-11-08 by zssherman

mgrover1, 2022-11-08:
@bdolan44 this is now available in the latest release of Py-ART

bdolan44, 2022-11-08:
Fantastic! I was just thinking about this functionality again recently. Thanks!