ARM-DOE/pyart #853: Plotting Issues

zzwei1, 2019-07-19:
I used the following Python2.7 script to plot data from NEXRAD. 

import matplotlib.pyplot as plt
import pyart
import numpy as np
filename = 'KTLX20160914_111932_V06'
radar = pyart.io.read_nexrad_archive(filename)
display = pyart.graph.RadarDisplay(radar)
fig = plt.figure(figsize=(6, 5))
ax = fig.add_subplot(111)
#radar = scan.open_pyart()
display = pyart.graph.RadarDisplay(radar)
display.plot('reflectivity',0,ax=ax,title='NEXRAD Reflectivity',vmin=-32,vmax=64,colorbar_label='')
display.plot_range_ring(radar.range['data'][-1]/1000.,ax=ax)
display.set_limits((-500,500),(-500,500), ax = ax)
plt.show()
/home/zzwei/桌面/1.png
[image]

While my figure looks like a little strange.I don't know how to fix it.And when I import pyart from IPython console,It shows:
 **You are using the Python ARM Radar Toolkit (Py-ART), an open source library for working with weather radar data. Py-ART is partly supported by the U.S. Department of Energy as part of the Atmospheric Radiation Measurement (ARM) Climate Research Facility, an Office of Science user facility.**
  **If you use this software to prepare a publication, please cite:
 JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119
/home/zzwei/.local/lib/python2.7/site-packages/pyart/graph/cm.py:104: FutureWarning: elementwise comparison failed; returning scalar instead, but in the future will perform elementwise comparison**
  **if 'red' in spec:
/home/zzwei/.local/lib/python2.7/site-packages/pyart/graph/cm_colorblind.py:32: FutureWarning: elementwise comparison failed; returning scalar instead, but in the future will perform elementwise comparison**
  **if 'red' in spec:
/home/zzwei/anaconda2/lib/python2.7/site-packages/numpy/lib/type_check.py:546: DeprecationWarning: np.asscalar(a) is deprecated since NumPy v1.16, use a.item() instead
  'a.item() instead', DeprecationWarning, stacklevel=1)
/home/zzwei/anaconda2/lib/python2.7/site-packages/numpy/lib/type_check.py:546: DeprecationWarning: np.asscalar(a) is deprecated since NumPy v1.16, use a.item() instead
  'a.item() instead', DeprecationWarning, stacklevel=1)**

I don't know whether it is related to my issue.It doen't seem like my data is broken.

zssherman, 2019-07-19:
@zzwei1 Hmm that is interesting. I don't see anything wrong with your plotting code, can you try a plot but set cmap='pyart_NWSRef' in the plot code? I want to see if maybe something is wrong with the colormap.

zzwei1, 2019-07-20:
That's awesome!
I have set cmap='pyart_NWSRef',and the figure seems like right.
[image]
Does it mean that there's something wrong with the default colormap?

zssherman, 2019-07-20:
@zzwei1 Sweet, yeah that helps narrow it down. I think something is wrong with the code for the default homeyer rainbow color map in pyart. I'll look into what could be causing this.

zzwei1, 2019-07-20:
> @zzwei1 Sweet, yeah that helps narrow it down. I think something is wrong with the code for the default homeyer rainbow color map in pyart. I'll look into what could be causing this.

Thank you very much!

zssherman, 2019-07-23:
@rcjackson Do you possibly have an idea of what could be causing this in the colorblind code? I looked through the cm_colorblind code and didn't really see anything. But this issue and #747 seem to have similar issues with the colormap.

rcjackson, 2019-07-23:
Is there an issue with float to integer conversion of the colormap? Maybe some versions of python the integer division in the loop is not functioning correctly.

zssherman, 2019-07-24:
@rcjackson I think you might be right, I'm still trying to pinpoint the issue, but with python 3.6 instead of python 2.7, I plotted the same data but the colormap worked.
[image]

zssherman, 2019-07-24:
#856 Fixes this issue for python 2.7 users.

zssherman, 2019-07-24:
Fixed, closing.

closed 2019-07-24 by zssherman