ARM-DOE/pyart #858: Issue Reading CF/Radial 2.0 Files

mpichault, 2019-07-29:
Hi,

I am experiencing issues trying to read LiDAR data through Py-Art. I am processing data from a Leosphere WindCube 400S and the output format is NETCDF4, based on the CF/Radial 2.0 (CF-1.7) conventions. Below is an example of the error message I am getting when trying to read the netcdf4 files. 

I am wondering whether PyArt is designed to read the Leosphere CFRadial 2 format files or if I need to convert them to another readable format? Forgive me if my question is trivial; I am only getting started with Python.

Thank you very much for your time and consideration.

Mathieu.

`  File "/anaconda3/envs/pyart/lib/python3.6/site-packages/pyart/io/auto_read.py", line 121, in read
    return read_cfradial(filename, **kwargs)    # CF/Radial

  File "/anaconda3/envs/pyart/lib/python3.6/site-packages/pyart/io/cfradial.py", line 149, in read_cfradial
    metadata[var] = str(netCDF4.chartostring(ncvars[var][:]))

  File "netCDF4/_netCDF4.pyx", line 6061, in netCDF4._netCDF4.chartostring

AttributeError: 'str' object has no attribute 'dtype' `

kmuehlbauer, 2019-07-29:
@mpichault First a disclaimer: I'm not a pyart-maintainer but part-time dev/user for several years now. 

Up to now, pyart can work with CfRadial1 up to CF-1.6 I think. For debugging the current error one source file would be necessary. Then it could be checked, if there can be worked around somehow or if you would need to convert to another format.

mpichault, 2019-07-29:
Hi @kmuehlbauer.  Thank you for your response. 
I have attached below an example of a source I have tried to process in vain.
Kind regards,
M 
[WLS400s-158_2019-07-26_02-14-25_ppi_122_75m.nc.zip](https://github.com/ARM-DOE/pyart/files/3440525/WLS400s-158_2019-07-26_02-14-25_ppi_122_75m.nc.zip)

kmuehlbauer, 2019-07-29:
@mpichault OK, from a first inspection I can reproduce the error. 

The error is due to the fact that Cf/Radial1 (CF-1.6) does expect a netcdf array of chars but in the file is a string. This could be handled by a try/except-clause.

But the next problem: there is no `time` variable/coordinate/dimension (and other expected variables) in the `root`-group of the file (according to Cf/Radial2 it is in the sweep group). I do not see a immediate solution without massive tampering with pyart/cfradial-reader, but pyart-devs might have a solution at hand.

If there is no quick solution at hand, your choices are to either convert your data to something pyart can read or to use other readers. But this depends on your workflow and what you want to do with the data.

scollis, 2019-07-29:
Thanks @kmuehlbauer ! Lets keep this issue open as I want to test some LIDAR data from ARM's Halo Photonics systems which are meant to be CF-Radial

meteoswiss-mdr, 2019-11-04:
Hi @scollis, all,

I had to read some data in CFRadial2 format so I created a reader compatible with Py-ART. Probably it has to be improved but you can find the current version here:
https://github.com/meteoswiss-mdr/pyart/blob/master/pyart/io/cfradial2.py

It may be used as a basis for proper CFRadial2 ingestion in Py-ART.

Greetings from Switzerland!

zssherman, 2019-11-04:
Thanks @meteoswiss-mdr ! I'll take a look.

zssherman, 2022-11-03:
This is being tackled in xradar, discussion can be moved here:

closed 2022-11-03 by zssherman