ARM-DOE/pyart #859: install error

cycle13, 2019-07-31:
I try to install pyart package in os with gcc-9.
There occur an error when the compile the c program.
I paste the information in following, could you give some hint? 
Thank you so much.
.......
building extension "io._sigmetfile" sources
building extension "io.nexrad_interpolate" sources
building data_files sources
build_src: building npy-pkg config files
running build_py
running build_ext
customize UnixCCompiler
sh: line 1: 10235 Segmentation fault: 11  'gcc' --version 2> /dev/null > '/var/folders/0f/2ky54s2j77sf0ynpdhy_3x1w0000gn/T/tmpfdq4at31'
customize UnixCCompiler using build_ext
building 'io._sigmetfile' extension
compiling C sources
C compiler: gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I//anaconda3/include -arch x86_64 -I//anaconda3/include -arch x86_64

compile options: '-I//anaconda3/lib/python3.7/site-packages/numpy/core/include -I//anaconda3/lib/python3.7/site-packages/numpy/core/include -I//anaconda3/include/python3.7m -c'
gcc: ./_sigmetfile.c
error: Command "gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I//anaconda3/include -arch x86_64 -I//anaconda3/include -arch x86_64 -I//anaconda3/lib/python3.7/site-packages/numpy/core/include -I//anaconda3/lib/python3.7/site-packages/numpy/core/include -I//anaconda3/include/python3.7m -c ./_sigmetfile.c -o build/temp.macosx-10.7-x86_64-3.7/_sigmetfile.o -MMD -MF build/temp.macosx-10.7-x86_64-3.7/_sigmetfile.o.d" failed with exit status -11

scollis, 2019-07-31:
What platform? 

-sent from a mobile device-

scollis, 2019-07-31:
Just about to go to bed, but did you install conda compilers 
https://anaconda.org/conda-forge/compilers

-sent from a mobile device-

zssherman, 2019-07-31:
@cycle13 Yeah I think it's a compiler issue too. It wouldn't hurt to update cython too, but I would try the compiler in the link @scollis gave.

zssherman, 2019-10-01:
@cycle13 Are you still having issues? If not I'll probably close this issue.

zssherman, 2019-11-27:
Closing for now, can be reopened if needed.

closed 2019-11-27 by zssherman