ARM-DOE/pyart #861: Issue with altitude and gridding when reading in Level3 Files

guidodev, 2019-08-02:
First, MANY THANKS to all the contributors to this project, what an amazing tool you've created.

Wanted to let you know I found an issue with radar site altitudes when reading in Level3 files. It appears that at least sometimes the site altitudes imported from the Level3 files are in feet, but pyart is interpreting them in meters. This tripped me up for awhile, because when gridding the data the radius of influence can miss the grid in the z dimension, leading to what appears to be "missing data" and "holes in the data" for no obvious reason.

I already have a workaround that suits my needs and am assuming this probably isn't a high priority fix. Just wanted to make the community aware in case someone else comes across this problem or if the devs want to fix it.

zssherman, 2019-08-02:
@guidodev Thank you for those words! 
If you believe that your changes would be a good addition to Py-ART, I would recommend a pull request. If not, do you have a specific case or example file that I can check this out with?

guidodev, 2019-08-02:
@zssherman 

My workaround would not be appropriate at all. The z-axis isn't important to me, so I'm just setting altitude to zero in each radar object.

Here's a sample file:

> import sys
> import pyart
> 
> radar = pyart.io.read_nexrad_level3('dhr_test_file.nids')
> 
> print(radar.altitude['data'])
> 

The result is:

> [3703.]

This is for Amarillo, which is about 3,700' elevation, and grid_from_radars seems to interpret this as meters instead of feet. Hope that helps!

[dhr_test_file.zip](https://github.com/ARM-DOE/pyart/files/3462979/dhr_test_file.zip)

dcedgren, 2020-05-19:
@zssherman 

This issue was reidentified in Issue #922. The fix to that issue also addresses this one, so I think you can close this issue as well.

zssherman, 2020-05-19:
@dcedgren Yep, good catch! I will close this one too. Much appreciated!

closed 2020-05-19 by zssherman