ARM-DOE/pyart #862: Option to use `plt.imshow` instead of `plt.pcolormesh` in `plot_grid`.

eshort0401, 2019-08-07:
I've noticed that the matplotlib function used to plot grids (as well as longitude/latitude slices) is `plt.pcolormesh`. 

https://github.com/ARM-DOE/pyart/blob/ceb4d545223d96208bd23e2e6bb74a63a906776f/pyart/graph/gridmapdisplay.py#L206

The default `shading='flat'` behaviour of this function shades grid-cells by treating grid values as the corners of each grid-cell, then averaging the corner values to decide how to shade that grid-cell. The other option, `shading='gourard'`, shades the grid-cell with a colour gradient based off the corner values, although this option currently [doesn't work](https://github.com/matplotlib/matplotlib/issues/8422).

I have found the averaging/interpolation behaviour of `pcolormesh` makes tasks like visually identifying maximum in data difficult - currently I need to do this to check to check whether my updraft identification method is working properly. I think a function like `plt.imshow` would better represent the actual values of the grid data - maybe there could be an option in `plot_grid` to use `plt.imshow` in place of `plt.pcolormesh`?

zssherman, 2019-08-12:
@eshort0401 Thank you for the suggestion! I will look into this.

zssherman, 2021-10-11:
Is now in pyart, closing.

closed 2021-10-11 by zssherman