ARM-DOE/pyart #866: Issues with CyLP

saralytle, 2019-09-10:
We're having issues getting cylp to run on jupyter notebooks - it immediately restarts the kernel when any phase processing is done, namely: 

 reproc_phase, kdp = pyart.correct.phase_proc_lp(radar,0., refl_field='reflectivity',
                                                fzl=13000.,
                                                self_const=60000.0,
                                                window_len=20,
                                                low_z=10.,high_z=53.)

We've created multiple environments, using the environment.yml file provided here, and all of our trials haven't succeeded in getting it to run. Sometimes a 'gcc' complier error will pop up, other times cylp appears to install successfully, but the kernel continues to restart whenever we try phase processing.

Any advice on what to try next or something to check?

scollis, 2019-09-10:
What machine?
Is this is for Japan I would reach out to Zach re running at ORNL.. 

Dr Scott Collis
Atmospheric Scientist, Argonne National Laboratory
Radar Instrument Translator, ARM User Facility
Head, Environmental Computing, Data, Sensing and GIS Department.
Senior Fellow, Northwestern Argonne Institute of Science and Engineering
Cell: 630 235 8025
GitHub: https://github.com/scollis
Oh My, a Blog: https://opensky.press/

mvanlierwalq, 2019-09-10:
I've pinged @zssherman on an issue that matches ours and also separately on slack. Yeah, this is for Japan. we were hoping to have a little more analysis done that what we showed at ASR and SLS.

zssherman, 2019-09-10:
@mvanlierwalq I responded to your message on the other issue, the compiler issue might be solved with recent cython version and using anaconda c compilers. I would take a look at this comment here:
https://github.com/coin-or/CyLP/issues/62#issuecomment-441334466
might also help if these are the compiler warnings your running into to. The main branch of CyLP should work with the different python 3 versions, so no need to downloads anybodys specific branch anymore, but if worse case that still isn't working, might want to default back to Jonathan's branch and use:
pip install git+https://github.com/jjhelmus/CyLP.git@py3

mvanlierwalq, 2019-09-10:
installing the condaforge c and cxx compilers, updating cython and reinstalling cylp from scratch solved the issue. Thanks for your help!

zssherman, 2019-10-01:
I added info in the README.rst and INSTALL.rst with a reference to the Anaconda compilers for using dependencies such as CyLP, closing for now.

closed 2019-10-01 by zssherman