ARM-DOE/pyart #870: #plan position indicator (PPI) radar display from .h5 doppler weather radar data using pyart.

Asishinuzuka, 2019-10-02:
How do i display plan position indicator (PPI) radar display from .h5 (hdf) doppler weather radar data using pyart and i have doppler weather data in binary with .dwr as extension, how to i convert it to hdf format?

zssherman, 2019-10-02:
@Asishinuzuka Once you have a radar object, you can use the code found here to plot a ppi, just subtract the map and use plot_ppi if you don't want a map. http://arm-doe.github.io/pyart/dev/auto_examples/plotting/plot_ppi_cartopy_with_rings.html
I'm not entirely sure how to convert that extension to hdf format, could always try one of the hdf readers in pyart found in the aux_io, otherwise someone else in the community might have encountered this before.

Asishinuzuka, 2019-10-02:
Thank you for the answer, and actually any format will work (not have to be in .h5 format) .netcdf or anything else, i just want to know what does the file contain and learn about it.

zssherman, 2019-10-02:
What kind of errors do you get when trying to read using pyart? 
You can also try h5py.File(filename, 'r')

Asishinuzuka, 2019-12-24:
so sorry for the late reply, the error i am getting is: 
> KeyError: 'time'
while plotting a PPI.

scollis, 2020-01-22:
Not an issue.. please use the google group https://groups.google.com/forum/#!forum/pyart-users

closed 2020-01-22 by scollis