ARM-DOE/pyart #871: Failed to read uf file

Kwonil-Kim, 2019-10-14:
Hello, I've tried to read uf file but I got a failure.
It seems the reason is that the number of bins varies by elevation angles, 

Here is the error I've got.
```
>>> radar = pyart.io.read_uf('/disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/site-packages/pyart/io/uf.py", line 181, in read_uf
    field_dic['data'] = ufile.get_field_data(uf_field_number)
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/site-packages/pyart/io/uffile.py", line 199, in get_field_data
    raw_data[i, :bins] = ray.field_raw_data[field_number]
ValueError: could not broadcast input array from shape (1200) into shape (1192)
```

The attached uf file has 1192 bins for the first three tilts and 1200 bins for higher tilts.
Please unzip the file before reading it.
[RDR_GRS_201803150900.zip](https://github.com/ARM-DOE/pyart/files/3722709/RDR_GRS_201803150900.zip)

Thanks in advance for your help.

zssherman, 2019-10-14:
@Kwonil-Kim Hmm, can you try:
radar = pyart.aux_io.read_radx(filename)
I wonder if maybe radx can read this file.

Kwonil-Kim, 2019-10-14:
@zssherman Thank you for the answer.
Should I install another software to use read_radx?

```
>>> pyart.aux_io.read_radx('/disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/site-packages/pyart/aux_io/radx.py", line 52, in read_radx
    '-outdir', head, '-outname', tail, '-f', filename])
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'RadxConvert': 'RadxConvert'
```

Kwonil-Kim, 2019-10-14:
@zssherman I apologize I didn't fully read your comment. Please allow me some time to install radx..

Kwonil-Kim, 2019-10-16:
@zssherman 
I was able to install Radx software. However, it returns the error as well (please see below).

```
>>> pyart.aux_io.read_radx('/disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf')
======================================================================
Program 'RadxConvert'
Run-time 2019/10/16 10:48:12.

Copyright (c) 1992 - 2019
University Corporation for Atmospheric Research (UCAR)
National Center for Atmospheric Research (NCAR)
Boulder, Colorado, USA.

Redistribution and use in source and binary forms, with
or without modification, are permitted provided that the following
conditions are met:

1) Redistributions of source code must retain the above copyright
notice, this list of conditions and the following disclaimer.

2) Redistributions in binary form must reproduce the above copyright
notice, this list of conditions and the following disclaimer in the
documentation and/or other materials provided with the distribution.

3) Neither the name of UCAR, NCAR nor the names of its contributors, if
any, may be used to endorse or promote products derived from this
software without specific prior written permission.

4) If the software is modified to produce derivative works, such modified
software should be clearly marked, so as not to confuse it with the
version available from UCAR.

======================================================================
ERROR - RadxConvert::_readFile
  path: /disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf
ERROR - RadxFile::readFromPath
  File format not recognized: /disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf
ERROR - GenericRadxFile::readFromPath
  File format not recognized: /disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf

ERROR - running RadxConvert
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/site-packages/pyart/aux_io/radx.py", line 52, in read_radx
    '-outdir', head, '-outname', tail, '-f', filename])
  File "/home/kwonil/.conda/envs/pyart/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['RadxConvert', '-const_ngates', '-outdir', '/home/kwonil/TEST/GRS/lrose-core-lrose-core-20190520', '-outname', 'tmph33nwwbb.nc', '-f', '/disk/STORAGE/OBS/Radar/MLTM/UF/201803/15/RDR_GRS_201803150900.uf']' returned non-zero exit status 255.

```

I can read the file without any issues if I try with rsl library in IDL or C.
When I change the nbins in the uf header (for all rays, all sweeps, and all fields) to 1200, then pyart can read it.
I think it would be great if you could consider making pyart to read this radar with this kind of scan strategy.