ARM-DOE/pyart #875: ERROR: file or package not found: pyart (missing __init__.py?)

cachocomadreja, 2019-10-25:
Hi! I've just installed pyart using recommended command:

**conda env create -f environment.yml**

It installs correctly, apparently, but then Python fails when importing pyart. I run:

**pytest --pyargs pyart**

And fails with the message in the title:

**ERROR: file or package not found: pyart (missing __init__.py?)**

What could be the problem? Thanks in advance.-

PS.: I had an environment created before that worked correctly until it broke somehow (?) so I made a fresh pyart installation and now this came up!

zssherman, 2019-10-25:
Hi @cachocomadreja , when pyart installs with environment.yml its from conda-forge. If you want to install pyart and use pytest on it, I would install pyart  with:
pip install -e .
after git clone the pyart repository. Also install pytest within the environment.

```
conda create -n pyart_env -c conda-forge python=3.7 numpy scipy netCDF4 pytest cartopy
source activate pyart_env
git clone https://github.com/ARM-DOE/pyart.git
cd pyart
pip install -e .
```
then pytest within that directory should work. conda forge install, you don't have a directory to unit tests on, atleast what I'm aware of.

cachocomadreja, 2019-10-25:
Thank you! Actually, I wasn't interested in using pytest. I just tried it to see what was wrong with my environment as it suddenly stopped working. 

What I did to make it work is repeating the installation process but from base instead of the new environment and it worked (which I wanted to avoid in the first place).

Thanks.-

zssherman, 2019-10-25:
Ah gotcha, thought the issue at first was trying to run pytest on it, but anyways glad to hear it works now.

scollis, 2020-01-22:
Sounds like it is working now.. closing

closed 2020-01-22 by scollis