ARM-DOE/pyart #882: [Enhancement] Accessing and editing radar.info()

aschueth, 2019-11-21:
I was digging through the docs on the radar class and I was wondering if it was possible and/or a desired feature to be able to pull out the full radar info as a dictionary and have it be editable like the metadata. As far as I understand, the current architecture does not support this. You can pull certain parts of the info as a dictionary using the private method _dic_info, but it only pulls the data, it does not enable the user to write data back into the radar object. I'm not sure what the pros and cons are to a feature like this, but it would be useful to me to add extra information.

scollis, 2020-01-22:
Thanks @aschueth .. By this do you mean to be able to edit the value in the radar metadata..

ie radar.metadata.update({'Somekey':value})?

radar info is a method
http://arm-doe.github.io/pyart-docs-travis/user_reference/generated/pyart.core.Radar.info.html#pyart.core.Radar.info

you need to track down where different parts are drawn from (eg ngates etc...)
you should be able to modify them and depending on the writer (eg CFRadial) it should write

aschueth, 2020-01-22:
Yeah edit a value, but also add new fields as well. For instance, with mobile radars, I would like to add a variable on deployment number.

Right now I have a sidecar csv file that I use to get information out of, but it makes my workflow a bit messy, so I was trying to streamline it such that all my information could be in the CFRadial file structure and could be accessed, modified and added on to using the metadata method. 

Would it be possible to have a mode flag in the method so you could read in the data (default mode='r'), but could also change it to write a dictionary with mode='w'? Unfortunately, there would have to be a lot of checks to make sure the dictionary is compatible?

scollis, 2020-01-22:
cool bananas... you can add this to the radar.metadata field now.. we do this routinely.. see:
https://github.com/EVS-ATMOS/cmac2.0/blob/master/cmac/cmac_radar.py#L319

where we basically strip out the metadata and add our own..

radar.metadata maps to the global headers of the CF-radial file so you can put anything you want in there...

aschueth, 2020-01-22:
Oh awesome, I wasn't aware of that feature.. Thanks for the information, I think this is exactly what I need!

closed 2020-01-22 by aschueth