ARM-DOE/pyart #884: About the paramters of ZPHI attenuation

sysumeteo, 2019-11-22:
what are the reference for the default parameters (a_coef/beta/c/d) of ZPHI atteuation ? I can not find those values in the reference of the ZPHI method (Gu et.al 2011)

zssherman, 2019-11-25:
@rcjackson Do you know the reference for the default coefficients that sysumeteo is referring to?

rcjackson, 2019-11-25:
It looks like the default parameters are grabbed from a function:

def _param_attzphi_table():
    """
    Defines the parameters of Z-Phi attenuation estimation at each frequency
    band.
    Returns
    -------
    param_att_dict : dict
        A dictionary with the coefficients at each band.
    """
    param_att_dict = dict()

    # S band:
    param_att_dict.update({'S': (0.02, 0.64884, 0.15917, 1.0804)})

    # C band:
    param_att_dict.update({'C': (0.08, 0.64884, 0.3, 1.0804)})

    # X band:
    param_att_dict.update({'X': (0.31916, 0.64884, 0.15917, 1.0804)})

    return param_att_dict

However, there is no reference for them. This was the new method added by MeteoSwiss. @meteoswiss-mdr , do you know where these values come from?

meteoswiss-mdr, 2019-11-26:
I do not recall exactly where they come from originally but they should be consistent with the article of Ryzhkov et al. 2014. They have been used internally at MeteoFrance and MeteoSwiss.

scollis, 2020-01-22:
@rcjackson when you get a moment let me know your thoughts on this...

rcjackson, 2020-01-22:
I suspect they used simulated DSDs from somewhere. I found a figure in Ryzhkov's paper showing their simulated values for alpha. The black dots represent 0 degrees C, the grey dots are 20 degrees C. It looks like the alpha values are in the ranges that would be expected. I don't really know of a way to verify the other values, which is troublesome for reproducibility. 
[image]

scollis, 2020-01-22:
Perhaps we need a constants section with this data in there and well referenced to various papers...

scollis, 2020-03-18:
@rcjackson can you do two things so we can close this out:
1) Add a notebooks showing application to X,C and S band in https://github.com/ARM-DOE/pyart/tree/master/notebooks (We will eventually binder this) .. Data don't have to be included but orderable (so XSapr, C-SAPR and NEXRAD)
2) beef up docs

scollis, 2020-03-18:
not the highest priority... Lets aim for a month or so.

meteoswiss-mdr, 2020-03-23:
Sorry I cannot be more helpful here. For sure I used Ryzhkov paper for most of the coefficients (derived from disdrometer data). Others come from empirical measurements performed at MeteoFrance. They were collected in internal reports from MeteoFrance.

gewitterblitz, 2021-02-12:
@scollis @rcjackson Any updates on this?

mo-sun-jin, 2022-01-25:
[http://chuvaproject.cptec.inpe.br/portal/pdf/relatorios/a2anexo3a7.pdf]

[image]

The coefficient values ​​appear to be referenced from the linked site papers.

The title of the final thesis is as in the image. There are also differences in the coefficient values ​​in the paper.

zssherman, 2022-02-02:
Thank @mo-sun-jin for the heads up, I'll take a look at the paper!

rcjackson, 2022-02-02:
We should modify the values and perhaps send out an email in pyart-users about this correction.

zssherman, 2022-02-02:
Seems the defaults are a reference to:
Ryzhkov et al. Potential Utilization of Specific Attenuation for Rainfall
Estimation, Mitigation of Partial Beam Blockage, and Radar Networking,
JAOT, 2014, 31, 599-619.

With discussion with rcjackson the values change between midlatitude vs tropical which could explain the difference in the Chuva campaign and the paper mentioned in Py-ART's references

scollis, 2022-02-03:
Are they hardset in the method? Can we make them keyword parameters? This is always going to be a challenge in Py-ART. Thank you for bringing this up Mo-Sun-Jin  From: Zach Sherman ***@***.***>Date: Wednesday, February 2, 2022 at 1:40 PMTo: ARM-DOE/pyart ***@***.***>Cc: Scott ***@***.***>, Mention ***@***.***>Subject: Re: [ARM-DOE/pyart] About the paramters of ZPHI attenuation correction (#884)Seems the defaults are a reference to:Ryzhkov et al. Potential Utilization of Specific Attenuation for RainfallEstimation, Mitigation of Partial Beam Blockage, and Radar Networking,JAOT, 2014, 31, 599-619.With discussion with rcjackson the values change between midlatitude vs tropical which could explain the difference in the Chuva campaign and the paper mentioned in Py-ART's references

zssherman, 2022-02-03:
@scollis Users can set the coefficients, those values are defaults that are used if not specified.

scollis, 2022-02-03:
Cool. Just catching up2nd Bobby’s suggestion  From: Zach Sherman ***@***.***>Date: Thursday, February 3, 2022 at 3:33 PMTo: ARM-DOE/pyart ***@***.***>Cc: Scott ***@***.***>, Mention ***@***.***>Subject: Re: [ARM-DOE/pyart] About the paramters of ZPHI attenuation correction ***@***.*** Users can set the coefficients, those values are defaults that are used if not specified.

zssherman, 2024-05-16:
Closing in favor of #1042

closed 2024-05-16 by zssherman