ARM-DOE/pyart #887: Support for NEXRAD RDA Build 19.0

dopplershift, 2019-12-21:
I wanted to give you all a heads up that, based on the [draft ICD](https://github.com/ARM-DOE/pyart/files/3991230/Draft_Baseline_2620002S_ICD_for_RDA_RPG_B19.pdf) for the next NEXRAD RDA build, there are some rather significant changes to message 31 coming. I think this table sums it up:

[image]

1. They're adding a moment CFP (Clutter Filter Power Removed). Looks like there's a lot of places you list out the possible moments, not to mention run through a fixed set of data block pointers.

2. They're also increasing ZDR to 16-bit storage (for 11-bit data). That would seem to run afoul of:
https://github.com/ARM-DOE/pyart/blob/f13a8d7ec42f485c1b0528abdaa448005d839d37/pyart/io/nexrad_level2.py#L534-L538
 FYI, the 8/16-bit word size is encoded in the data block header.

I don't have any sample data, but thought you all might appreciate some warning.

zssherman, 2019-12-21:
Thanks @dopplershift for the heads up! I'll look into some changes to adapt to this. I appreciate the warning!

scollis, 2019-12-21:
Absolutely appreciated! Do you have a contact in the ROC we could ask? 

 

https://github.com/ARM-DOE/pyart/blob/f13a8d7ec42f485c1b0528abdaa448005d839d37/pyart/io/nexrad_level2.py#L534-L538

dopplershift, 2019-12-23:
@scollis You mean for some sample data? not really. However, it turns out that one of the testbeds (FOP1) is transmitting data on the operational feed, and I've confirmed with MetPy that this data has the additional moment and 2 byte ZDR data. I've attached a [sample file](https://github.com/ARM-DOE/pyart/files/3993231/Level2_FOP1_20191223_003655.ar2v.gz)
, but you can also get more from the AWS bucket or from our THREDDS servers.

scollis, 2020-01-22:
@zssherman can you look at the example file and start ensuring Py-ART will continue to work..

zssherman, 2020-04-03:
Done in #903 closing.

closed 2020-04-03 by zssherman