ARM-DOE/pyart #888: Speed up ‘Nearest’ method for mapping and add it into ‘map_gates_to_grid’？

sysumeteo, 2019-12-25:
‘Nearest’ method is quite useful for mapping FHC data, but unfortunately  was supported only by   algorithm 'map_to_grid ' which is several times slower than 'map_gates_to_grid'. Is there any plan to speed up 'Nearest' method and add it into ‘map_gates_to_grid’?
BTW:  Function 'grid_from_radars' use algorithm 'map_gates_to_grid'  by default, but with no suggestion about its incompatible with 'Nearest' method in the LATEST DOC. That  keeps confusing me until i found annotation in the code. I think adding some suggestion in the doc would be better.

zssherman, 2019-12-28:
Hi @sysumeteo Nearest should be in map_gates_to_grid, https://github.com/ARM-DOE/pyart/blob/master/pyart/map/gates_to_grid.py#L170 if its not working for some reason, then there possibly is a bug then. I will take a look at the docs too.

sysumeteo, 2019-12-30:
@zssherman  My inattention for the function ! Actually map_gates_to_grid support  'Nearest' method, and run as fast as map_gate_to_grid should be.But another problem is that ,'Nearest' method perform quite different between 'map_gate_to_grid' and 'map_to_grid'. 
My code and plots are showed below:
 The first FH plot is  corresponding to 'Nearest' method of 'map_to_grid',while the second one is corresponding to 'Nearest' method of 'map_gate_to grid' and the third plot is for 'Barnes2' of 'map_gate_to grid' .

        grid=pyart.map.grid_from_radars((radar1,),grid_shape=(nz,nx,ny),grid_limits=((min_z,max_z),(min_x,max_x),(min_y,max_y)),weighting_function='Barnes2')       
        #grid_FH=pyart.map.grid_from_radars((radar1,),grid_shape=(nz,nx,ny),grid_limits=((min_z,max_z),(min_x,max_x),(min_y,max_y)),
        #       fields=['FH'],gridding_algo='map_to_grid',weighting_function='Nearest')        
        grid_FH=pyart.map.grid_from_radars((radar1,),grid_shape=(nz,nx,ny),grid_limits=((min_z,max_z),(min_x,max_x),(min_y,max_y)),
               fields=['FH'],weighting_function='Nearest')

[image]
[image]
[image]

zssherman, 2019-12-30:
Hmm, I'll have to look into that. I'm not sure why the two plots are giving different results for Nearest.

zssherman, 2020-05-22:
Fixed in #934 closing.

closed 2020-05-22 by zssherman