ARM-DOE/pyart #894: pyart.graph.RadarMapDisplay AttributeError: 'GeoAxesSubplot'

salbenson, 2020-01-15:
Hi, 
I am just starting to use py-art.  I am working with pyart.graph.  I am using this website as my reference for pyart.graph.  http://arm-doe.github.io/pyart-docs-travis/user_reference/graph.html  On this page there is RadarMapDisplay and RadarMapDisplayBasemap.  I am guessing that these are the same thing, except RadarMapDisplay uses cartopy instead of basemap.  Am I right?  

Here are my python lines using RadarMapDisplay:
display=pyart.graph.RadarMapDisplay(radar)
fig=plt.figure()
ax=fig.add_subplot(111)
display.plot_ppi_map('reflectivity',0)
plt.show()

When I run this I am getting this error:  AttributeError: 'GeoAxesSubplot' object has no attribute '_hold' 

Any advice on how to debug this error?  Is this a py-art error or a cartopy error?  Below is a copy of all the error output from spyder.  Thank you for your help.  
Regards, 
Sally

In [2]: runfile('/uufs/chpc.utah.edu/common/home/u0079358/matchproj-python-master/scripts_sb/plot_three_panel_gridmapdisplay.py', wdir='/uufs/chpc.utah.edu/common/home/u0079358/matchproj-python-master/scripts_sb')
Reloaded modules: __mp_main__
Traceback (most recent call last):

  File "/uufs/chpc.utah.edu/common/home/u0079358/matchproj-python-master/scripts_sb/plot_three_panel_gridmapdisplay.py", line 109, in <module>
    display.plot_ppi_map('reflectivity',0)

  File "/uufs/chpc.utah.edu/common/home/u0079358/software/pkg/miniconda3/lib/python3.7/site-packages/pyart/graph/radarmapdisplay.py", line 283, in plot_ppi_map
    norm=norm, transform=self.grid_projection)

  File "/uufs/chpc.utah.edu/common/home/u0079358/software/pkg/miniconda3/lib/python3.7/site-packages/cartopy/mpl/geoaxes.py", line 1451, in pcolormesh
    result = self._pcolormesh_patched(*args, **kwargs)

  File "/uufs/chpc.utah.edu/common/home/u0079358/software/pkg/miniconda3/lib/python3.7/site-packages/cartopy/mpl/geoaxes.py", line 1469, in _pcolormesh_patched
    if not self._hold:

AttributeError: 'GeoAxesSubplot' object has no attribute '_hold'

zssherman, 2020-01-15:
Hi @salbenson, this was an issue in an earlier Cartopy, [here](https://github.com/SciTools/cartopy/issues/1120) what Cartopy version do you have?

salbenson, 2020-01-15:
Hi, I got cartopy using conda install cartopy.  When I type conda list this is the information I get about cartopy.

cartopy                   0.16.0           py37hfa13621_0

zssherman, 2020-01-15:
I believe the issue was fixed in Cartopy 0.17, I would update your Cartopy and it should fix the issue.

salbenson, 2020-01-15:
Thanks for your help.  Updating cartopy to 0.17 fixed the GeoAxesSubplot error.  Now I am getting another error.  I have pasted it below.  Do I need to update "shapely"? 

File "/uufs/chpc.utah.edu/common/home/u0079358/software/pkg/miniconda3/lib/python3.7/site-packages/shapely/geometry/linestring.py", line 115, in xy
    return self.coords.xy

AttributeError: 'list' object has no attribute 'xy'

zssherman, 2020-01-15:
I'll do a new release for Py-ART, if you git clone pyart and install the github repo from source this will fix the issue, but I haven't done the conda release yet. I'll do that now.

zssherman, 2020-01-15:
The conda release is done, might take 10-15 minutes or so before it appears:
https://anaconda.org/conda-forge/arm_pyart/files

salbenson, 2020-01-15:
Great, Thanks!  I'll give it a try.

salbenson, 2020-01-16:
I got the new conda release of py-art.  That fixed the issue.  Thank you very much!

zssherman, 2020-01-16:
Great to hear! Closing.

closed 2020-01-16 by zssherman