ARM-DOE/pyart #895: ODIM h5 file stays open after reading when using `pyart.aux_io.read_odim_h5`

vlouf, 2020-01-16:
TL;DR: The function `pyart.aux_io.read_odim_h5` does not close the hdf5 files after reading the file, whether it read it successfully or raised an error. I have worked on a fix I'll send a pull request.

H5 files are kept open open in memory when read with the odim_h5 reader. Any attempt to read the file again fails and throws an OSError. The solution is to do a `gc.collect()`. 

Moreover, if an error is raised while reading a file with `pyart.aux_io.read_odim_h5`, then that file can't be accessed again until you shutdown the python kernel or did a `gc.collect()`. To reproduce this error you can, for example: 
- Open a cf/radial file with `pyart.aux_io.read_odim_h5`, it will throw an expected error: `KeyError: "Unable to open object (object 'what' doesn't exist)"`
- Try to open the same file with the correct function `pyart.io.read_cfradial`, you get: 

```python
pyart/io/cfradial.py in read_cfradial(filename, field_names, additional_metadata, file_field_names, exclude_fields, include_fields, delay_field_loading, **kwargs)
    125 
    126     # read the data
--> 127     ncobj = netCDF4.Dataset(filename)
    128     ncvars = ncobj.variables
    129 

netCDF4/_netCDF4.pyx in netCDF4._netCDF4.Dataset.__init__()

netCDF4/_netCDF4.pyx in netCDF4._netCDF4._ensure_nc_success()

OSError: [Errno -101] NetCDF: HDF error
```

scollis, 2020-01-16:
Thanks @vlouf ..
Assigning to @rcjackson

scollis, 2020-01-16:
This looks fairly easy to fix.. When you have a chance @rcjackson can you investigate

scollis, 2020-01-16:
Oh.. never mind.. I see @vlouf has a PR :D

vlouf, 2020-01-16:
Also: you should consider using the `with` statement for netCDF files in `pyart.io.read_cfradial`. Because in the case that an error is raised while reading the file, it will stay open in memory. However, because of the `delay_field_loading` argument` L325 of  `pyart.io.read_cfradial` you can't do that

scollis, 2020-01-16:
@zssherman can you check out @vlouf suggestion.. either you or he can raise a new issue

zssherman, 2020-01-16:
Yeah, the nc.close happens later in the code, so if there is a failure, I could see it staying in memory and 'with' would avoid this. There might have been a reason why 'with' wasn't use originally, but I'll have to dig deeper into that. Who wrote the original cfradial code?

kmuehlbauer, 2020-01-16:
>  Who wrote the original cfradial code?

@jjhelmus , at least this is what the commits say.

scollis, 2020-01-22:
Ok, I believe #896 closes this... If not please reopen

closed 2020-01-22 by scollis