ARM-DOE/pyart #900: Limited Projections for plot_grid

lgaudet, 2020-01-31:
Hi! I have run into an issue with plotting a Cartopy map with the 'LambertConformal' projection when using the 'plot_grid' function. While the gridded radar data is plotted, the background map (e.g., state boundaries) does not show up. 
[image]
However, when I change the projection to 'PlateCarree', I can see the gridded radar data and state boundaries in that specific projection. Is 'PlateCarree' the only projection that works with 'plot_grid'? I'd be happy to provide more information, if needed. Also, this is the first issue I've opened on GitHub, so please forgive me if I am not properly posting this.
[image]

zssherman, 2020-01-31:
Hi @lgaudet , welcome to the GitHub community! What version of cartopy are you using? I'll try to see if I can reproduce the issue.

lgaudet, 2020-01-31:
@zssherman Thank you! I'm using version 0.17.0 of cartopy. If you need the preceding code that grabs radar data from multiple radar sites via AWS for the 'grid_from_radars' below, let me know.
```
grids = pyart.map.grid_from_radars(radars,
                        grid_origin=(lat_central, lon_central),
                        grid_shape=(46,251,251),
                        grid_limits=((0, 20000.), (-1*out_reach, out_reach), (-1*out_reach, out_reach)),
                        fields=['reflectivity'],
                        gridding_algo="map_gates_to_grid",
                        weighting_function='BARNES')
#projection = ccrs.LambertConformal()
projection = ccrs.PlateCarree()

display = pyart.graph.GridMapDisplay(grids)

fig = plt.figure(figsize=(8,8))
ax1 = fig.add_axes(projection=projection)
display.plot_grid('reflectivity', level=1,
                  vmin=-8, vmax=64,
                  projection=projection,
                  cmap = pyart.graph.cm.NWSRef)
display.plot_crosshairs(lon=-74, lat=43., linewidth=4)
plt.show()
```

zssherman, 2020-01-31:
Hmm, I was able to get it to work with RadarMapDisplay, but not GridMapDisplay, so I was able to reproduce the issue. I'll take a look in the GridMapDisplay code, see if there is a bug somewhere.

zssherman, 2020-01-31:
@lgaudet Seems since platecarre is already what the data is in, the projection shows, but if going to lambert, the transform is needed as a parameter to transform from platecarree to lambert. Try adding in your plot_grid function:
display.plot_grid('reflectivity', level=1,
                  vmin=-8, vmax=64,
                  projection=projection, transform=ccrs.PlateCarree()
                  cmap = pyart.graph.cm.NWSRef)
For lambert projection add lat and lon as so:
# Setting projection, figure size, and panel sizes.
projection = ccrs.LambertConformal(
    central_latitude=grids.origin_latitude['data'][0],
    central_longitude=grids.origin_longitude['data'][0])
let me know if this works

lgaudet, 2020-01-31:
@zssherman That worked! Thank you so much for looking into this, I should have known I needed a transform.

zssherman, 2020-01-31:
Awesome glad to hear! I'll close this issue for now.

closed 2020-01-31 by zssherman