ARM-DOE/pyart #901: Adding multiple cartopy maps to display.plot_ppi_map

tjturnage, 2020-02-02:
Hello,
I've been using your display.plot_ppi_map method and can successfully add a single map along with the kwargs that I care most about. However, I often find myself wanting to add multiple shapefile layers, some of which are not available as Natural Earth Features (for example, damage survey maps).  It's true I could merge layers ahead of time, but that would mean less flexibility and more work for the end user.

I therefore request that you add functionality to iteratively plot multiple cartopy map layers in display.plot_ppi_map. This would likely require another argument/option that looks for one of the following:

1) A list of variables representing shapefile paths
2) A dictionary containing shapefile paths and respective user-defined kwargs for each shapefile

Sample methods for what I'm describing are at:
[https://github.com/tjturnage/resources/blob/master/gis_layers.py](https://github.com/tjturnage/resources/blob/master/gis_layers.py)

Thanks for considering!

TJ

scollis, 2020-02-02:
Nice idea. Making it easier to add layers is a great feature add.

I’m on my phone right now so hard to point you to the right place, but, as a stopgap all the axes etc are stored in the display object. 

-sent from a mobile device-

tjturnage, 2020-02-02:
Thanks for the quick reply! Yes, please point me to the right place when you have the time.

Thanks again,
TJ

scollis, 2020-02-02:
@zssherman , @rcjackson can you show @tjturnage how to grab axes from radarmapdisplay and add layers? I’m up to my elbows in grease working on the mountain bike

rcjackson, 2020-02-02:
For now, the RadarMapDisplay stores the geoaxes instance as the .ax property of the RadarMapDisplay object. On this, you should be able to do all of the standard cartopy layer routines that you need. For example, to add coastlines:

display = pyart.graph.RadarMapDisplay(radar)
display.plot_ppi_map('reflectivity', sweep=0)
ax = display.ax
ax.coastlines()
# Other layers....

tjturnage, 2020-02-20:
Never got back to you, but your workaround works great. Thanks!

TJ

closed 2020-02-20 by tjturnage