ARM-DOE/pyart #904: display.plot_ppi_map lat_lines and lon_lines when set to None

allenea, 2020-02-12:
Hi,

I was trying to plot some radar images without having gridlines. However any time that I try to do it I get partial gridlines anyway. Is there a bug or am I doing something wrong.

FYI
arm_pyart                 1.11.0           py37h3b54f70_0    conda-forge
cartopy                   0.17.0          py37h95120c7_1007    conda-forge

Here is with the gridlines:

        fig = plt.figure(figsize=(8,8), dpi=100)

        # Grid data into geographic coordinates
        display = pyart.graph.RadarMapDisplay(radar)

        # Setting projection and ploting the second tilt
        projection = cartopy.crs.LambertConformal(central_latitude=radar.latitude['data'][0],
                                           central_longitude=radar.longitude['data'][0])
        # Display
        display.plot_ppi_map('reflectivity', 0, vmin=0, vmax=60,\
                             min_lon=-77, max_lon=-73.8, min_lat=37, max_lat=40.25,\
                             lon_lines=np.arange(-180, 0, 1), resolution='10m',\
                             lat_lines=np.arange(0, 90, 1), projection=projection,\
                             fig=fig, lat_0=radar.latitude['data'][0],\
                             lon_0=radar.longitude['data'][0],\
                             title_flag=True,\
                             title=display.generate_title("reflectivity", 0).split("Deg.")[0]+\
                             "Deg. "+timestamp+"\n NEXRAD Equivalent Reflectivity (dBZ)",                    
                             colorbar_flag=True,\
                             colorbar_label="Equivalent Reflectivity Factor (dBZ)",\
                             cmap=cmap, norm=norm)

        plt.show()
        plt.close()

[image]

Compared to without gridlines (None):

        fig = plt.figure(figsize=(8,8), dpi=100)

        # Grid data into geographic coordinates
        display = pyart.graph.RadarMapDisplay(radar)

        # Setting projection and ploting the second tilt
        projection = cartopy.crs.LambertConformal(central_latitude=radar.latitude['data'][0],
                                           central_longitude=radar.longitude['data'][0])
        # Display
        display.plot_ppi_map('reflectivity', 0, vmin=0, vmax=60,\
                             min_lon=-77, max_lon=-73.8, min_lat=37, max_lat=40.25,\
                             lon_lines=None, resolution='10m',\
                             lat_lines=None, projection=projection,\
                             fig=fig, lat_0=radar.latitude['data'][0],\
                             lon_0=radar.longitude['data'][0],\
                             title_flag=True,\
                             title=display.generate_title("reflectivity", 0).split("Deg.")[0]+\
                             "Deg. "+timestamp+"\n NEXRAD Equivalent Reflectivity (dBZ)",                    
                             colorbar_flag=True,\
                             colorbar_label="Equivalent Reflectivity Factor (dBZ)",\
                             cmap=cmap, norm=norm)

        plt.show()
        plt.close()

[image]

scollis, 2020-02-12:
Thank you for the very detailed issue report! I'll assign @zssherman to investigate shortly.. In the meanwhile can you try setting embelish =False (just realize we have a spelling mistake there!)

allenea, 2020-02-12:
Yep :) 
In the meantime this works. Thanks for looking into it.

Eric

mgrover1, 2022-11-09:
Closed with #1297 , @allenea this is available with the most recent Py-ART release

closed 2022-11-09 by mgrover1