ARM-DOE/pyart #905: PyGIDeprecationWarning when I import pyart

jayavanth, 2020-02-18:
I get a bunch of PyGIDeprecationWarnings when I import pyart in `ipython`. It doesn't let me type anything and it just throws these errors at me

```
In [1]: import pyart

## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.
##
## If you use this software to prepare a publication, please cite:
##
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119

/usr/local/lib/python3.6/dist-packages/matplotlib/backends/backend_gtk3.py:45: DeprecationWarning: Gdk.Cursor.new is deprecated
  cursors.MOVE          : Gdk.Cursor.new(Gdk.CursorType.FLEUR),

In [2]: /usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
/usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
/usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
/usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
In [2]: 

In [2]: /usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
/usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
/usr/lib/python3/dist-packages/gi/overrides/GLib.py:724: PyGIDeprecationWarning: Calling io_add_watch without priority as second argument is deprecated
  PyGIDeprecationWarning)
In [2]: 

```

scollis, 2020-02-18:
Thanks @jayavanth I don't use IPython so have never seen this.. @zssherman can you try and do an import of Py-ART into IPython?

zssherman, 2020-02-18:
@jayavanth @scollis I couldn't reproduce the error with an Ipython in python 3.6, are you in anyway able to print out package versions? Maybe try updating matplotlib or glib?

jayavanth, 2020-02-18:
@zssherman @scollis Thanks for the help! I figured it out. It was my mistake. After installation, I was using IPython in the pyart/ source directory and the python env was picking up the library from the current directory. I changed the directory and everything works fine.

closed 2020-02-18 by jayavanth

zssherman, 2020-02-18:
No problem, glad you were able to get it to work!

jayavanth, 2020-02-19:
Ah that was a solution to a different problem. I fixed this one by upgrading IPython from 5.0 to 7.12.0. Either way, it's fixed :)