ARM-DOE/pyart #906: Optional parameter "doc" and "freq_band"

jegehrin, 2020-02-19:
Dear developers of attenuation,

I realized that the parameter "doc" that is described as optional is actually required, otherwise there is an error on line 189 of "phase_proc.py" in: 
` gate_end = min(gate_end, len(ranges) - doc).`

Also if the frequency in radar. instruments_parameters is out of the range [2e9, 12e9], then there is an error in the warning of line 725 in attenuation.py:    
```
 warn("Radar frequency out of range. "
         + "Coefficients only applied to S, C or X band. "
         + freq_band + " band coefficients will be used.")
```
I think that `freq_band_aux` should be used in the warning instead, is that correct ?

Thank you,

Josué

zssherman, 2020-02-19:
I believe I see it, doc has a default value in phase_proc.py, but in attenuation some of the functions default it to None, so it feeds in None which breaks the code, I'll see what I can do for that. And I believe you are also right on the freq_band_aux code too that it should be freq_band_aux. Thanks for catching that!

zssherman, 2020-03-18:
Fixed in #907 closing

closed 2020-03-18 by zssherman