ARM-DOE/pyart #908: Py-ART bug?

simaberson, 2020-02-19:
We are using Py-ART for our
    project and recently updated to python3.  In the nexrad_level2.py, there are the lines
    
    _> grep get_msg29 nexrad_level2.py
             new_pos = _get_msg29_from_buf(buf, pos, dic)
    def _get_msg29_from_buf(pos, dic):
    
    It's calling with three arguments but receives only two.  All the other get_msg* lines have three arguments, so we modified the last line, and it works fine.

We installed Py-ART using anaconda.  We have arm_pyart-1.11.0-py37hc1659b7_0
    
    Sim

scollis, 2020-02-19:
Thanks @simaberson .. Assigning to @zssherman

zssherman, 2020-02-19:
Yeah looks like buf is not even used in that function, so it can be removed from the call. I'll do a change shortly. Thanks for catching that!

zssherman, 2020-03-18:
Fixed in #909 closing.

closed 2020-03-18 by zssherman