ARM-DOE/pyart #922: Nexrad Level 3 - Elevation in Feet, not meters

dcedgren, 2020-04-01:
For Product #176 (DPR) files the altitude is read into Pyart as meters, not feet as per the Nexrad documentation.

I suspect this may also be an issue with other products. I will confirm that this is an issue with other products and, if so, convert from feet to meters when the unit setting is meters (as is the default).

zssherman, 2020-04-01:
Hi @dcedgren I'll submit a pull request, so essentially the altitude needs to be multiplied by 0.3048?

dcedgren, 2020-04-01:
Correct.

But only if the user's units are set to meters. If they've changed them to feet we wouldn't want to convert.

Testing will also need to be changed. Including for the PR I'm working on. I intentionally made this wrong.

zssherman, 2020-04-01:
Ah gotcha, hmm, ok I'll close my PR, got to find a way to check if there units are already in meters.

zssherman, 2020-04-01:
@dcedgren It currently gets the height from self.prod_descr, which is unpacked from buffer.

dcedgren, 2020-04-01:
As far as I'm aware the Nexrad file will always be in feet. (Maybe someday we'll eventually manage to switch entirely to metric...)

BUT, it looks like the units might be switched by the user in default_config.py. If someone had switched this to feet instead I assume we wouldn't want to convert to meters. Unless this is never meant to be changed. You tell me.

We could easily check units at line 118 of nexradl3_read.py.

if altitude['units'] == 'meters':

zssherman, 2020-04-01:
Yeah, there users can change the config to their liking for field naming etc, but in terms of units I would believe meters is what we want too keep @scollis your thoughts? The gridding code and more do everything in meters.

zssherman, 2020-05-18:
Done in #924 merging

closed 2020-05-18 by zssherman