ARM-DOE/pyart #928: Chunked L2 Data

Craytor, 2020-04-10:
I was wondering if pyart supports using chunked L2 data from, for instance, NOAA Big Data (https://unidata-nexrad-level2-chunks.s3.amazonaws.com).

If it doesn't "natively" support it, is there a way to convert these chunks into a version that pyart would accept? Trying to learn more about radar.

dopplershift, 2020-04-13:
Is there a reason you want to look at the individual chunks rather than the full volumes available at https://noaa-nexrad-level2.s3.amazonaws.com/index.html ? To be clear, the full volumes are just an ordered concatenation of the individual chunks you mentioned above.

Craytor, 2020-04-14:
Yeah, the issue is noaa-nexrad-level2 data is not real-time and is often delayed 8-10 minutes for the full volume scan to complete. I have since solved this issue by creating my own slice chunks to read.

closed 2020-04-14 by Craytor

grwise1, 2021-01-31:
I was able to use this solution [https://github.com/Unidata/MetPy/issues/1470](url) to open the chunks in Metpy Level2file format. However, I am disappointed to be missing out on pyart's extensive dealiasing and filtering capabilities. If anyone does have any suggestions on this, please let me know. Receiving the chunks data is ideal for my application, especially with the ability to capture SAILS scans, mid volume.

dopplershift, 2021-02-03:
@grwise1 You shouldn't need to make any changes, with MetPy 1.0 things should "just work"--if they don't, please do [open an issue](https://github.com/Unidata/MetPy/issues/new).

I don't know about pyart's routines specifically, but one of the reasons chunks are difficult to deal with is that there is important information (e.g. number of scans, nominal scan angles, etc.) at the start of the volume. Also, most tools are geared to work with entire volumes because until recently whole volumes were all you were able to get. Honestly, I'm kind of the opinion that crucial algorithms that you need aren't available, it's not really true that "chunks data is ideal".

grwise1, 2021-02-03:
Hi Ryan,

I appreciate all your feedback. I just want to try to clarify why I am approaching this from such an angle. If I wait for the entire volume scan to complete, I will be minutes delayed from when the initial base scan occurred. If I instead capture the chunk data, I can receive the base scan without waiting for the entire volume to complete. (for example: Running code that captures only the entire volume, I will be about 4-5 minutes behind Radarscope's 0.5° data.)

I have had success gathering, plotting and publishing base data from the chunks. Heck, since the chunks come in 60° sweeps, I've even been able to live animate the scan process. This is all just in the drive to receive and plot the data sooner than typical applications. 

I may attempt to concatenate these chunks as they come in, in order to satisfy pyart...

Thanks again.

unphased, 2024-12-28:
I have found in testing that `s3://unidata-nexrad-level2-chunks/KMLB/955/20241228-095013-012-I` is the 12th sub-chunk of `s3://noaa-nexrad-level2/2024/12/28/KMLB/KMLB20241228_095013_V06`. It is my understanding that each chunk is individually bzip2-compressed in some way that isn't directly unpackable with bzip2 command line tool. 

I hope to know if it is possible to prepare an analogous file that pyART may be able to consume, which has the bzip2 decompression precomputed (e.g. an uncompressed volume record). Currently I believe I could potentially prepare that by reusing some code on each of the chunks and then maybe concatenating them or reading each chunk. 

You can see my attempt here 

[image]

I feel like the -S ("start"?) chunk missing an MSG31 (message type 31?) indicates this load method is not suited. 

I'd like to be able to have the flexibility of isolating bzip2 handling and radar information processing across disparate processes.