ARM-DOE/pyart #930: Weighting Function discrepancy between map_gates_to_grid and map_to_grid

cptmedic, 2020-05-05:
Greetings,

There appears to be a slight discrepancy in map_gates_to_grid when using nearest as the weighting function when compared to map_to_grid at the same grid_limits. 

map_gates_to_grid with nearest weighting (producing weird rays):
[image]

map_to_grid with nearest weighting (data is seamless):
[image]

These are two examples with the same grid_limit the only difference is the function used.

Also discussed at https://groups.google.com/forum/#!topic/pyart-users/bQJ_1JuAYAg

zssherman, 2020-05-05:
@cptmedic Thanks again for submitting this! We will take a look at the code and see what could be causing this.

zssherman, 2020-05-22:
Hi @cptmedic, this should now be fixed in #934

closed 2020-05-22 by zssherman