ARM-DOE/pyart #931: Bug in zphi retrieval

joshua-wx, 2020-05-13:
Hi,
In the zphi attenuation retrieval (https://github.com/ARM-DOE/pyart/blob/master/pyart/correct/attenuation.py) on line 257, a base 10 power is taken to calculate the 'self_cons_number'. The formula for this technique requires the use of a natural exponent (See Wang et al 2019 equation 4, attached). I've tested this extensively and it corrects an issue where the pyart implementation gives specific attenuation an order of magnitude too high.

The multiplier in your implementation of equation 4 also looks wrong, but maybe you're scaling beta to compensate? It's 0.23 in the paper, and 0.1 in the code. beta should be around 0.62 for S band.

[Wang etal 2019.pdf](https://github.com/ARM-DOE/pyart/files/4619610/Wang.etal.2019.pdf)

Cheers!
Joshua

joshua-wx, 2020-05-13:
So it looks like the two reference provided for ZPHI are two different formulations... one used a base10 exponential (Gu et al) and the other used a natural exponent (Ryzhkov), with different coefficients in the exponent term to adjust for this change. Suggest a note is added to the docs explicitly saying the formulation is from Gu et al

zssherman, 2020-05-15:
Hi @joshua-wx I'll add a note, and after discussion with @scollis will look into the other formulation. Thank you for finding this!

PR #1579 (2024-05-15): DOC: Update the attenuation docs to reflect where the formula comes from.
Doc change was suggested in #931

- [x] Closes #931
- [x] Documentation reflects changes
closed 2024-05-15 by zssherman